arduino/Arduino #2526: Duplicate pages "Arduino Build Process" need to be consolidated and updated

michaelshiloh, 2015-01-01:
http://arduino.cc/en/Hacking/BuildProcess and https://code.google.com/p/arduino/wiki/BuildProcess both describe the build process. 

1) There should be only one such page and the duplicate should be removed

2) The contents should be consolidated and updated for Arduino 1.0 and 1.5

michaelshiloh, 2015-01-12:
On 01/12/2015 05:48 AM, Arturo Guadalupi wrote:> Hi Michael!
> Look @ this:
> https://github.com/arduino/Arduino/issues/2526
> 
> You should close this issue. The google code pages are very old and they
> should be all edited to please visit arduiino.cc because as i said the
> contents are very old!

My response:

I will study this. If something needs to be done then it should be done before the issue is closed.

I believe Arturo is saying:

"The google code pages are very old and they
should be all edited to SAY please visit arduiino.cc"

I will look at the original issue and if this seems to be the appropriate action, I will take that action before I close the issue. This really is part of a global issue and the solution should be carefully thought out and not performed piecemeal and randomly as it is found.

Perhaps I should consult with whoever is coordinating all of our content, since this is a major issue. We have lots of very old material which is worth keeping for historical reasons and for reference but should be labeled as such with pointers to the up-to-date information.

q2dg, 2015-03-01:
+1

Maybe adding most updated page (arduino.cc's, it seems) to the Github wiki??

agdl, 2015-06-12:
I added a statement (in google code) that redirects to the updated documentation

https://code.google.com/p/arduino/wiki/BuildProcess?ts=1434111835&updated=BuildProcess

closed 2015-06-12 by agdl

q2dg, 2015-09-06:
Documentation that can be found in https://github.com/arduino/Arduino/wiki/Building-Arduino is NOT updated as it states at beginning:

*"The process the Arduino environment (1.0.x) uses to build a sketch.
Please note that this page has been migrated from google code and has to be updated"*

On the other hand, is http://arduino.cc/en/Hacking/BuildProcess more updated than former page, or not? Having two places with the same aim is a lot confusing!!

Please:
1.-Reopen this issue
2.-Select and update one page and erase the other

agdl, 2015-09-07:
@q2dg this can remain closed because is a duplicate of https://github.com/arduino/Arduino/issues/3653

q2dg, 2015-09-07:
Ah, I see. Sorry.