arduino/Arduino #2527: 1.5.8 add new core via boards.txt

NicoHood, 2015-01-02:
I cannot include my new core with the boards.txt

Thats my sketchbook path:
Arduino\sketchbook\hardware\HID\avr\cores\hid

arduino:arduino works fine. But not my custom core.

```
##############################################################

leonardo_extended.name=Arduino Leonardo Extended HID-Core
leonardo_extended.vid.0=0x2341
leonardo_extended.pid.0=0x0036
leonardo_extended.vid.1=0x2341
leonardo_extended.pid.1=0x8036
leonardo_extended.upload.tool=arduino:avrdude
leonardo_extended.upload.protocol=avr109
leonardo_extended.upload.maximum_size=28672
leonardo_extended.upload.maximum_data_size=2560
leonardo_extended.upload.speed=57600
leonardo_extended.upload.disable_flushing=true
leonardo_extended.upload.use_1200bps_touch=true
leonardo_extended.upload.wait_for_upload_port=true

leonardo_extended.bootloader.tool=arduino:avrdude
leonardo_extended.bootloader.low_fuses=0xff
leonardo_extended.bootloader.high_fuses=0xd8
leonardo_extended.bootloader.extended_fuses=0xcb
leonardo_extended.bootloader.file=arduino:caterina/Caterina-leonardo_extended.hex
leonardo_extended.bootloader.unlock_bits=0x3F
leonardo_extended.bootloader.lock_bits=0x2F

leonardo_extended.build.mcu=atmega32u4
leonardo_extended.build.f_cpu=16000000L
leonardo_extended.build.vid=0x2341
leonardo_extended.build.pid=0x8036
leonardo_extended.build.usb_product="Arduino Leonardo"
leonardo_extended.build.board=AVR_LEONARDO
#leonardo_extended.build.core=arduino:arduino
#leonardo_extended.build.core=HID:hid
leonardo_extended.build.core=hid
leonardo_extended.build.variant=leonardo_hid
leonardo_extended.build.extra_flags={build.usb_flags}
```

Error:
```
Third-party platform.txt does not define compiler.path. Please report this to the third-party hardware maintainer.

processing.app.debug.RunnerException
	at processing.app.debug.Compiler.getCommandCompilerCPP(Compiler.java:593)
	at processing.app.debug.Compiler.compileFiles(Compiler.java:271)
	at processing.app.debug.Compiler.compileSketch(Compiler.java:638)
	at processing.app.debug.Compiler.compile(Compiler.java:121)
	at processing.app.Sketch.build(Sketch.java:1588)
	at processing.app.Sketch.build(Sketch.java:1508)
	at processing.app.Editor$DefaultRunHandler.run(Editor.java:1912)
	at java.lang.Thread.run(Unknown Source)
Caused by: java.lang.NullPointerException
	at processing.app.helpers.StringReplacer.replaceFromMapping(StringReplacer.java:97)
	at processing.app.helpers.StringReplacer.replaceFromMapping(StringReplacer.java:89)
	at processing.app.helpers.StringReplacer.formatAndSplit(StringReplacer.java:37)
	at processing.app.debug.Compiler.getCommandCompilerCPP(Compiler.java:591)
	... 7 more
```

ffissore, 2015-01-02:
Are you running the IDE or using the command line? If the latter, can you provide it as well?

MacroMachines, 2015-01-02:
That was confusing.. my name is nico too.  I thought I emailed myself.

NicoHood, 2015-01-02:
I am using the IDE 1.5.8 for windows in normal Window mode. The bug also happens with 1.6 rc1.

Here is an example. Keep in mind that the boards.txt referes to arduino:hid at the moment and that ou have to copy the hid folder to the Arduino folder because of that bug. Normally i would expect that you can copy the whole hardware thing to your sketchbook, rename arduino:hid to hid and it will work. But it doesnt.

https://github.com/NicoHood/HID/tree/Official-Integration

NicoHood, 2015-01-02:
Is there a way to include the pins_Arduino.h from the original source of a Leonardo from a custom variants folder? For example if i want to create something like the Micro, use the same pin layout but only change the Led behavior. What path do I have to use for this?

NicoHood, 2015-01-02:
Is there also a way to define new Keywords.txt for a core like for a library? If not should I open this as new issue?

ffissore, 2015-01-05:
Found it: you're defining a custom core for building sketches for this board, but you're not defining it: `leonardo_extended.build.core=hid`

@cmaglie I think [replacing compiler.path](https://github.com/arduino/Arduino/blob/ide-1.5.x/arduino-core/src/processing/app/debug/Compiler.java#L462-L469) alone does not suffice. We should fail more gracefully when platform is missing: do we have enough info to load a default/fallback one or should we just throw a RunnerException?

@NicoHood As for pins_arduino.h, AFAIK no, but you can define a pins_arduino in folder `HOME/Arduino/hardware/HID/avr` with content
```
#include "../../variants/leonardo/pins_arduino.h"
```

NicoHood, 2015-01-05:
I dont get your first sentence. I am defining it but i am not defining it? What? Am i doing something wrong and what should i do to fix it?

I found a solution for the inclusion:
```
// include the standard leonardo board definition file
#include <../../variants/leonardo/pins_arduino.h>
```

ffissore, 2015-01-05:
Right, sorry, the first *defining* was *assigning*. You're assigning a custom core called `hid` to the board but then you are not defining that core (compiler, tools locations and params and so on): in essence, you're missing a platform.txt

NicoHood, 2015-01-05:
my platform.txt is mostly empty. May you please help me to assign the correct compiler definitions etc? I just want to compile that new cor with the standard settings. Cant we make that the default option so its easier to add custom cores?

Is it then also possible to access a custom core from another Hardware file? Like if you have a special HID core (like me) and want to create a custom board, that also uses this hid core.

ffissore, 2015-01-05:
Leave `leonardo_extended.build.core` set to `arduino:arduino` and you'll have all the tools set up

NicoHood, 2015-01-05:
hm? i want to use the tools, sure. But of course i want to use the new core i made!?

ffissore, 2015-01-08:
Ok, so: if you want to use a totally new core and platform, also copy platform.txt from one of choice (avr in this case) and modify it appropriately.
A fix to the NullPointerException you notified is under way

NicoHood, 2015-01-09:
Working now but I dont like the idea. I'd rather prefer not to copy paste this stuff and use the default settings for new cores. And if that's technically not possible, please leave a simple notice about how to create new cores like you wrote me, ffisscore. Close if you think this issue is finished. Thanks :)

Here is my fix if someone is interested:
https://github.com/NicoHood/HID/commit/044e855c2e98861290d17bdcace09613f10ede93

NicoHood, 2015-01-09:
One more question:
How are new board definitions added to the ide? I dont see any logic in that. My HoodLoader is above the defaut stuff and my HID project is below. (HoodLoader not on the picture) https://github.com/NicoHood/HID/blob/Official-Integration/board.png

How can i push up this entry? I tried to change the names but since they are HID and HOO.. it shouldnt be too different?

NicoHood, 2015-01-09:
If i try to add the usb compile errors now i get
```
avr-g++: fatal error: no input files
compilation terminated.
Error compiling.
```

I also know where the problem is located. The usb product name needs to have a space in its name. No idea why and if this is the only bug workaround but it works.

HoodLoader2atmega16u2.build.usb_product=HoodLoader[space]2

cmaglie, 2015-01-27:
@NicoHood 
the "space" problem may be due to the missing quotes? the config line should look like that:

```
HoodLoader2atmega16u2.build.usb_product="HoodLoader 2"
```

please note the quotes around the name.
May you try check if this fixes your problem?

NicoHood, 2015-01-27:
No. There are quotes (missed it in my post) and the problem occurs when there is NO space. thats the weird thing, you wouldnt expect. the string needs a space to work

cmaglie, 2015-01-27:
May you link a boards.txt to reproduce the problem?

NicoHood, 2015-02-17:
As i said: just cross out the space in the leonardo file. Then the errors will occur as described above.

```
leonardo.name=Arduino Leonardo
leonardo.vid.0=0x2341
leonardo.pid.0=0x0036
leonardo.vid.1=0x2341
leonardo.pid.1=0x8036
leonardo.upload.tool=avrdude
leonardo.upload.protocol=avr109
leonardo.upload.maximum_size=28672
leonardo.upload.maximum_data_size=2560
leonardo.upload.speed=57600
leonardo.upload.disable_flushing=true
leonardo.upload.use_1200bps_touch=true
leonardo.upload.wait_for_upload_port=true

leonardo.bootloader.tool=avrdude
leonardo.bootloader.low_fuses=0xff
leonardo.bootloader.high_fuses=0xd8
leonardo.bootloader.extended_fuses=0xcb
leonardo.bootloader.file=caterina/Caterina-Leonardo.hex
leonardo.bootloader.unlock_bits=0x3F
leonardo.bootloader.lock_bits=0x2F

leonardo.build.mcu=atmega32u4
leonardo.build.f_cpu=16000000L
leonardo.build.vid=0x2341
leonardo.build.pid=0x8036
leonardo.build.usb_product="ArduinoLeonardo"
leonardo.build.board=AVR_LEONARDO
leonardo.build.core=arduino
leonardo.build.variant=leonardo
leonardo.build.extra_flags={build.usb_flags}
```

```
`avr-g++: fatal error: no input files
compilation terminated.
Error compiling.
```

cmaglie, 2015-02-17:
@NicoHood 
I can't reproduce the issue, if I remove the space it still compiles!
May be OS dependent? On which OS are you running it?

NicoHood, 2015-02-17:
Happens under windows 7 and 8

projectgus, 2015-02-26:
I ran into this issue just now testing our Freetronics board profile updates for 1.6.0 on Windows 7. Then I went back and re-tested using the same method @NicoHood used here, removing a space from the Arduino Leonardo product name.

Verbose log output when name is set to ArduinoLeonardo (ie no space):

```
Arduino: 1.6.0 (Windows 7), Board: "Arduino Leonardo"

Build options changed, rebuilding all

C:\Program Files (x86)\Arduino/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10600 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8036 -DUSB_MANUFACTURER= -DUSB_PRODUCT="ArduinoLeonardo" -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\leonardo C:\Users\gus\AppData\Local\Temp\build6165764808731065034.tmp\sketch_feb26a.cpp -o C:\Users\gus\AppData\Local\Temp\build6165764808731065034.tmp\sketch_feb26a.cpp.o 
avr-g++: error: Files: No such file or directory
avr-g++: error: (x86)Arduinohardwarearduinoavrcoresarduino -IC:Program: No such file or directory
avr-g++: error: Files: No such file or directory
avr-g++: error: (x86)Arduinohardwarearduinoavrvariantsleonardo C:\Users\gus\AppData\Local\Temp\build6165764808731065034.tmp\sketch_feb26a.cpp -o C:\Users\gus\AppData\Local\Temp\build6165764808731065034.tmp\sketch_feb26a.cpp.o: No such file or directory
avr-g++: fatal error: no input files
compilation terminated.
Error compiling.
```

When name is set normally (with space):

```
C:\Program Files (x86)\Arduino/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10600 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8036 -DUSB_MANUFACTURER= -DUSB_PRODUCT="Arduino Leonardo" -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\leonardo C:\Users\gus\AppData\Local\Temp\build8841861027842601922.tmp\sketch_feb26a.cpp -o C:\Users\gus\AppData\Local\Temp\build8841861027842601922.tmp\sketch_feb26a.cpp.o 
```
(... and compilation works fine.)

The generated command line strings look identical, which makes me think this is a bug in the StringReplacer.quotedSplit method ? Something is being quoted incorrectly under some circumstances.

I haven't had time to investigate further yet.

ffissore, 2015-09-29:
I think we can close this one: latest IDE is 1.6.5 (and upcoming 1.6.6 has hundreds of closed issues); @NicoHood is going on with the development of his hid core (see lots of issues here and at https://github.com/arduino/arduino-builder)

NicoHood, 2015-09-29:
Tested once again and seems to be fixed.
Updated my files finally XD

closed 2015-09-29 by NicoHood