arduino/Arduino #2529: app/build.xml need source 1.6

MauroMombelli, 2015-01-02:
please change line 85 of app/build.xml to use sources 1.6 (inted of 1.5) or the build will fail because of the use of diamon operator not supported..

ffissore, 2015-01-05:
The diamond operator is not used through out the source code and the IDE compiles fine with the current settings. What's the error message you're getting?

MauroMombelli, 2015-01-05:
it's not me, see http://forum.arduino.cc/index.php?topic=153592.msg2024388#msg2024388 (warining, italian language ahead)

ffissore, 2015-01-07:
Closing and handling this on the forum then. Thanks for forum link

closed 2015-01-07 by ffissore

ffissore, 2015-01-07:
Indeed user is trying autocomplete branch https://github.com/CriativaSoft/Arduino/tree/ide-1.5.x-autocomplete. We can't support it ATM