arduino/Arduino #2530: Attiny13 BUG Compilation

Camaard, 2015-01-02:
Hello.
I use Arduino Uno and I have it working well and without errors. But I decided to program using Uno microcontroller attini13 (sketch ISP). To do this, I used different files cores Attigny. But even when compiling errors occur:
Third-party platform.txt does not define compiler.path. Please report this to the third-party hardware maintainer.

processing.app.debug.RunnerException
	at processing.app.debug.Compiler.getCommandCompilerCPP(Compiler.java:597)
	at processing.app.debug.Compiler.compileFiles(Compiler.java:275)
	at processing.app.debug.Compiler.compileSketch(Compiler.java:642)
	at processing.app.debug.Compiler.compile(Compiler.java:125)
	at processing.app.Sketch.build(Sketch.java:1589)
	at processing.app.Sketch.build(Sketch.java:1509)
	at processing.app.Editor$DefaultRunHandler.run(Editor.java:1915)
	at java.lang.Thread.run(Unknown Source)
Caused by: java.lang.NullPointerException
	at processing.app.helpers.StringReplacer.replaceFromMapping(StringReplacer.java:97)
	at processing.app.helpers.StringReplacer.replaceFromMapping(StringReplacer.java:89)
	at processing.app.helpers.StringReplacer.formatAndSplit(StringReplacer.java:37)
	at processing.app.debug.Compiler.getCommandCompilerCPP(Compiler.java:595)
	... 7 more

tell me where to get the working files for attini13?
thank you

NicoHood, 2015-02-09:
You might try this: https://github.com/damellis/attiny/pull/44

ffissore, 2015-05-29:
Attiny core available via boards manager. Support available at https://github.com/damellis/attiny

closed 2015-05-29 by ffissore