arduino/Arduino #2532: Arduino IDE on High DPI

timhannah1988, 2015-01-03:
I have recently purchased a Microsoft Surface Pro 3 which has a High DPI screen. I installed the IDE and loaded it for the first time and it appears that the gui doesn't support DPI scaling as the menus and text is tiny and unreadable. I am having to use the device with a lower resolution so that I can use the software properly. 

Is there any support i am missing or is it simply not supported yet. If the latter please can this be considered on future builds

j40climb, 2015-01-15:
I'd like to see a fix for this as well. I have the same problem with my Samsung Ativ Book 9+. With all the new high DPI displays, this will become more of an issue.

PaulStoffregen, 2015-01-15:
This seems more like a Java question.  Arduino's IDE uses the Java JRE to do all those menus, so the bigger question is whether there's some way to tell the JRE to render them with more pixels?

j40climb, 2015-01-15:
Where is the UI code in the repro? Im knew to this project, but I'm willing
to dig in and look for a fix.

PaulStoffregen, 2015-01-16:
> Where is the UI code in the repro?

Here:

https://github.com/arduino/Arduino/tree/ide-1.5.x/app/src/processing/app

Please be aware there are 4 branches.  Do NOT use the default "master" branch, because that's the code for version 1.0.x, which is being phased out.  If you use that code, you'll have to redo everything to get it accepted.

Work within the "ide-1.5.x" branch, which is the latest version.

5YHgB, 2015-03-05:
[image]
I have this problem too. On my Lenovo Yoga 2 Pro the interface looks very small. Please fix.

PaulStoffregen, 2015-03-18:
If anyone is still following this issue and cares about Arduino gaining graphics and font scaling, please test the build on pull request #2776

choidamdam, 2015-08-19:
Any update on this? I'm using two windows laptops with HiDPI (one for work, one for home) displays and this is very annoying issue. 

Maybe we have to swich from Swing to JavaFX which support HiDPI better. (http://news.kynosarges.org/2015/06/29/javafx-dpi-scaling-fixed/)

ffissore, 2015-08-19:
JavaFX and Swing are two totally different beasts. They can integrate with each other, but it's more about rewriting the IDE that switching

Sembazuru, 2015-11-14:
I also value this issue. I can work-around the editing area by using 18pt font, but that only works in the actual code window. The Library and Board managers are almost unusable due to the small font being used on my high DPI monitor. Windows magnifier helps, but that is a kludge not a fix.

For reference, I'm on Windows10 64bit, and still see this behavior on IDE v 1.6.6.

thoughtfix, 2015-11-17:
+1  I am using the same trick as Sembazuru by boosting my font size, but it's still a miserable experience trying to code from a Surface Book.

mkstephenson, 2015-12-31:
I have the same problem, got a 4k 27" screen and you really have to squint to see any menu text

cmaglie, 2016-01-02:
I'm closing this one, please let's continue the discussion on #4376.

closed 2016-01-02 by cmaglie

roman-ku, 2017-02-27:
@thoughtfix Same story here on a surface book.

spacedSparking, 2018-01-23:
[image]

This is a life saver! Adjusting the scaling factor to 200% worked like a charm. I am currently running `ant run` within the `/Arduino/build/` directory from the terminal. For the total git noob here, is it possible to install/make the High DPI Arduino IDE as an bona fide application instead of following the [build instructions](https://github.com/processing/processing/wiki/Build-Instructions)?

cmaglie, 2018-01-23:
Sure, you can download the latest version here:

https://www.arduino.cc/en/Main/Software

delucaa, 2018-02-03:
On my surface pro 4 using the 1.8.5 version in the code window using the size 18 you can see the characters well but the window settings, libraries, you see small while in additional URLs the text is not seen almost at all
[image]

spacedSparking, 2018-02-03:
@delucaa Have you tried rebooting? I noticed the same issue before a restart, but everything looks properly scaled for me now.

delucaa, 2018-02-03:
yes but nothing ... I also tried to reinstall the program but nothing ..

cmaglie, 2018-02-05:
@delucaa please open a new issue, this should be an easy fix.