arduino/Arduino #2541: "efuse" memory type not defined for part "ATMEGA8"

rigelinorion, 2015-01-08:
Open for edit the file C:\Program Files\Arduino\hardware\arduino\avr\platforms.txt
this is standard section uploader/programmers tools

#==============================================================================
tools.avrdude.cmd.path={runtime.ide.path}/hardware/tools/avr/bin/avrdude
tools.avrdude.config.path={runtime.ide.path}/hardware/tools/avr/etc/avrdude.conf
tools.avrdude.cmd.path.linux={runtime.ide.path}/hardware/tools/avrdude
tools.avrdude.config.path.linux={runtime.ide.path}/hardware/tools/avrdude.conf

tools.avrdude.upload.params.verbose=-v -v -v -v
tools.avrdude.upload.params.quiet=-q -q
tools.avrdude.upload.pattern="{cmd.path}" "-C{config.path}" {upload.verbose} -p{build.mcu} -c{upload.protocol} -P{serial.port} -b{upload.speed} -D "-Uflash:w:{build.path}/{build.project_name}.hex:i"

tools.avrdude.program.params.verbose=-v -v -v -v
tools.avrdude.program.params.quiet=-q -q
tools.avrdude.program.pattern="{cmd.path}" "-C{config.path}" {program.verbose} -p{build.mcu} -c{protocol} {program.extra_params} "-Uflash:w:{build.path}/{build.project_name}.hex:i"

tools.avrdude.erase.params.verbose=-v -v -v -v
tools.avrdude.erase.params.quiet=-q -q
tools.avrdude.erase.pattern="{cmd.path}" "-C{config.path}" {erase.verbose} -p{build.mcu} -c{protocol} {program.extra_params} -e -Uefuse:w:{bootloader.extended_fuses}:m -Ulock:w:{bootloader.unlock_bits}:m -Uhfuse:w:{bootloader.high_fuses}:m -Ulfuse:w:{bootloader.low_fuses}:m

tools.avrdude.bootloader.params.verbose=-v -v -v -v
tools.avrdude.bootloader.params.quiet=-q -q
tools.avrdude.bootloader.pattern="{cmd.path}" "-C{config.path}" {bootloader.verbose} -p{build.mcu} -c{protocol} {program.extra_params} "-Uflash:w:{runtime.ide.path}/hardware/arduino/avr/bootloaders/{bootloader.file}:i" -Ulock:w:{bootloader.lock_bits}:m
#==============================================================================

create a new tool called, for example, avrdude1
simply copy section tools.avrdude and name it tools.avrdude1
the trick is that we remove switch "-Uefuse:w:{bootloader.extended_fuses}:m" from "tools.avrdude1.erase.pattern="

#=============================================================================
tools.avrdude1.cmd.path={runtime.ide.path}/hardware/tools/avr/bin/avrdude
tools.avrdude1.config.path={runtime.ide.path}/hardware/tools/avr/etc/avrdude.conf
tools.avrdude1.cmd.path.linux={runtime.ide.path}/hardware/tools/avrdude
tools.avrdude1.config.path.linux={runtime.ide.path}/hardware/tools/avrdude.conf

tools.avrdude1.upload.params.verbose=-v -v -v -v
tools.avrdude1.upload.params.quiet=-q -q
tools.avrdude1.upload.pattern="{cmd.path}" "-C{config.path}" {upload.verbose} -p{build.mcu} -c{upload.protocol} -P{serial.port} -b{upload.speed} -D "-Uflash:w:{build.path}/{build.project_name}.hex:i"

tools.avrdude1.program.params.verbose=-v -v -v -v
tools.avrdude1.program.params.quiet=-q -q
tools.avrdude1.program.pattern="{cmd.path}" "-C{config.path}" {program.verbose} -p{build.mcu} -c{protocol} {program.extra_params} "-Uflash:w:{build.path}/{build.project_name}.hex:i"

tools.avrdude1.erase.params.verbose=-v -v -v -v
tools.avrdude1.erase.params.quiet=-q -q

#this line was edited
tools.avrdude1.erase.pattern="{cmd.path}" "-C{config.path}" {erase.verbose} -p{build.mcu} -c{protocol} {program.extra_params} -e -Ulock:w:{bootloader.unlock_bits}:m -Uhfuse:w:{bootloader.high_fuses}:m -Ulfuse:w:{bootloader.low_fuses}:m

tools.avrdude1.bootloader.params.verbose=-v -v -v -v
tools.avrdude1.bootloader.params.quiet=-q -q
tools.avrdude1.bootloader.pattern="{cmd.path}" "-C{config.path}" {bootloader.verbose} -p{build.mcu} -c{protocol} {program.extra_params} "-Uflash:w:{runtime.ide.path}/hardware/arduino/avr/bootloaders/{bootloader.file}:i" -Ulock:w:{bootloader.lock_bits}:m
#==============================================================================

Here is an example of the use of the tool avrdude1 in boards.txt file

atmega8.name=ATmega8 OPTIBOOT
atmega8.upload.protocol=arduino
atmega8.upload.maximum_size=7168
atmega8.upload.speed=115200
atmega8.upload.tool=avrdude1
atmega8.bootloader.low_fuses=0xbf
atmega8.bootloader.high_fuses=0xCC
atmega8.bootloader.path=atmega8
atmega8.bootloader.file=optiboot/optiboot_atmega8.hex
atmega8.bootloader.unlock_bits=0x3F
atmega8.bootloader.lock_bits=0x0F
atmega8.bootloader.tool=avrdude1
atmega8.build.mcu=atmega8
atmega8.build.f_cpu=16000000L
atmega8.build.core=arduino
atmega8.build.variant=standard

closed 2015-01-08 by rigelinorion

d89443005, 2015-03-08:
I just encountered the problem after updating my Arduino IDE to the latest 1.6.0. I followed your instructions and succeeded in uploading to ATMega8 again! Thank you so much!

Manigandandmx, 2015-03-14:
Thanks you very much i successfully bootloaded my chips because of your advice.

rigelinorion, 2015-03-15:
Thanks. I'm glad I could help.

nirgos, 2015-08-30:
Thank you so much! This worked for me.

aakrutii, 2016-10-27:
**i have solved this problem check this link below** 

https://www.youtube.com/watch?v=2MMYnIFaXv4   

you will get the solution here

ramirovelazco, 2016-12-08:
This worked for me, with atmega8-16PU
Thanks!