arduino/Arduino #2547: Library file dependent on sketch file not compiling in Arduino IDE

wolfv6, 2015-01-09:
Please change Arduino IDE so it can compile library files that depend on sketch files.

Usually project files depend on library files.
But sometimes its useful to have dependences in the other direction.
In the following example, a library file depends on a sketch file.
Here is the example's directory structure:
```
sketch1\
	sketch1.ino
	CustomClass.h

sketch2\
	sketch2.ino
	CustomClass.h
	
Arduino\libraries\lib_folder\
	LibClass.h                        << LibClass.h depends on CustomClass.h
```
Each sketch has a CustomClass.h file.
The code within the customClass files is different for each sketch.
LibClass.h includes CustomClass.h
Downloaded example code here: [libClass_depend_on_sketchClass.zip](http://forum.arduino.cc/index.php?action=dlattach;topic=279427.0;attach=108978)

To compile sketch1.ino on Arduino 1.0.6 IDE, the user needs to move the corresponding CustomClass.h into lib_folder.
To compile sketch2.ino, the user needs to move that CustomClass.h into lib_folder.
Moving files in and out of libraries is dangerous and requires deleting Arduino's AppData\Local\Temp\build folder.
All the CustomClass files are moved in and out of the library every time all the sketchX.ino programs are regression tested.
Regression testing each versions of sketchX.ino will become impractical as more versions of sketchX.ino are added.
And instructing library users to move CustomClasses into the library is not very user friendly.

If CustomClass.h is not moved into the lib_folder, Arduino 1.0.6 IDE does not find the CustomClass.h file.
Here is the verbose error message:
```
Arduino: 1.0.6 + Td: 1.20 (Windows 7), Board: "Teensy 2.0"
C:\Program Files (x86)\Arduino\hardware\tools\avr\bin\avr-g++ -c -g -Os -Wall -fno-exceptions -ffunction-sections -fdata-sections -mmcu=atmega32u4 -DF_CPU=16000000L -MMD -DUSB_VID=null -DUSB_PID=null -DARDUINO=106 -DTEENSYDUINO=120 -felide-constructors -std=c++0x -DUSB_HID -DLAYOUT_US_ENGLISH -IC:\Program Files (x86)\Arduino\hardware\teensy\cores\teensy -IC:\Users\wolf\Documents\Arduino\libraries\lib_folder C:\Users\wolf\AppData\Local\Temp\build5396932314268139837.tmp\main.cpp -o C:\Users\wolf\AppData\Local\Temp\build5396932314268139837.tmp\main.cpp.o 

In file included from main.cpp:1:
C:\Users\wolf\Documents\Arduino\libraries\lib_folder/LibClass.h:4:25: warning: CustomClass.h: No such file or directory
In file included from main.cpp:1:
C:\Users\wolf\Documents\Arduino\libraries\lib_folder/LibClass.h:9: error: 'CustomClass' does not name a type
C:\Users\wolf\Documents\Arduino\libraries\lib_folder/LibClass.h: In member function 'void LibClass::fn()':
C:\Users\wolf\Documents\Arduino\libraries\lib_folder/LibClass.h:11: error: 'customObj' was not declared in this scope
```
Please change Arduino IDE so it can compile library files that depend on sketch files.

crteensy, 2015-12-01:
This request is basically what is described at https://www.arduino.cc/en/Hacking/BuildProcess in the Build Process paragraph, first sentence. And I'd love to see it implemented/working. Many libraries could benefit from this!

lmihalkovic, 2015-12-26:
It seems #15 is ahead on this topic.

wolfv6, 2016-05-11:
Thanks @lmihalkovic.

I no longer need this feature.  But it would still be useful for others.

This issue can be closed.

matthijskooijman, 2016-05-11:
I am indeed closing this, in favour of the relevant arduino-builder issue.

closed 2016-05-11 by matthijskooijman