arduino/Arduino #2549: wifi design files are wrong

jmiller712, 2015-01-11:
the WIFI Shield design files <arduino-wifi-shield-reference-design.zip>  do not match the wifi shield which now uses the HDG204 IC not the HDG104.

agdl, 2015-01-12:
@jmiller712 updated thank you!

jmiller712, 2015-01-12:
Arturo Guadalupi <notifications@github.com>

Hi Arturo.

Thanks, that was quick, but it's still not right?

Question 1
---------------
The new schematic still doesnt show the correct pinout for  HDG204?
The HDG104 and HDG204 use different pins for the antenna.
(Look close at earlier WIFI PCB around pins 4 & 5 of HDG104)

Question 2
---------------
The .brd file doesn't match the pcb?
Could you the updated the brd file too, ideally it would match the new PCB.
<Arduino Wifi shield v022-REV3.1.brd>

Question 3
--------------
Eagle says the brd and sch files are inconsistent?
Looking at the brd and sch files, Eagle says --
"Board and schematic are not consistent!"
"No forward-/backannotation will be performed!"
"Use the ERC command to get a detailed error report."

Thank you for seeing my feedback,  I didn't expect anyone would even notice it.
I really like the Arduino boards and all that you can do with them.

Thanks,  Jon Miller (805)564-3119

agdl, 2015-01-15:
Ops you are right I will upload the right files ASAP. I confused the two archives. I'm waiting for the right one.

agdl, 2015-09-01:
@jmiller712 I'm sorry, but since the shield is marked as retired I have to close this issue.

closed 2015-09-01 by agdl