arduino/Arduino #2553: faults in arduino uno schematic

tiefnig, 2015-01-17:
Hello,

I don't want to sound like a know it all and to be honest I have never really reported a bug. However I want to support this project in any way I can, so here are the errors I found:

1) fuse F1 - wrong symbol
2) wire drawn to long
3) +5V is labeled twice

Hope I'm right about this. 

[image]

bobc, 2015-01-17:
I wouldn't say those are errors, just cosmetic changes. It's clear what the design intent is.

nmt89, 2015-01-17:
yeah, that's true. But still, I thought it would be worth reporting...

cmaglie, 2015-07-01:
Thanks for the report, but we are not going to change the schematics (everyone has their own taste...).

closed 2015-07-01 by cmaglie