arduino/Arduino #2555: Editor 1.0.6 - suggestions

satelliteman, 2015-01-18:
Hi Arduino - Cool product - Ive just made my first automated RC car with GPS and it works sweet.
Now , what I want to discuss is the developing environment for creating the scripts . 
Im an Industrial electrician and have for 30 years been programming PLC's for automation , so it was relatively easy to come to grips with the editor . 

However - 2 things that really stuck out for me as issues ...(1) being very important.
1/ The colour used ( black ) on blue bacground for the TABS ...You cannot see these at all if you are in bad light or in the sun . - Terrible choice in colors ... when selected , the name of the tabs could be white text against black background etc .. not dark against dark . I have been debugging in the daylight and darkness ..you cannot read the tab names or know what is selected . 
2/ The terminology of UPLOAD and DOWNLOAD ...  - for 30 years I have  known major brands to Download code to the controller ... You UPLOAD to uplift or extract data from the controller . For sending the script to the controller , it is downloading .IMHO , the button should say Download. . 
3/Preferences - I would like to chose whether I want to open Previously opened script on starting Arduino app. - Check box please.
4/Perhaps the text color & background colors could have been an option for the preferences settings ?
 
Just suggestions , but hopefully someone will do something with them ..Thanks .

matthijskooijman, 2015-01-18:
Thanks for your suggestions. I'll respond to each one.

Regarding colors, I think you can change them in preferences.txt. It's a bit of a hassle, but perhaps it would help you.

Regarding upload vs download, I completely disagree. That is, it might be that convention is to use "download to the board", but Arduino aims at novice users, which are not burdened by convention. When look at the terms themselves, upload is often used for sending something away from your computer, e.g. to the intertnet. Hence using "upload to the board" makes sense, it names things from the perspective of your PC, which is where the Arduino IDE is running. Also, changing terminology now will instantly cause all existing documentation and tutorials to become out of date and even confusing, which is another argument for keeping things as they are. In any case, the best place for discussing these kinds of changes is the mailing list (see the contact use page on the Arduino site).

Reopening the previously open script sounds like a good, well-defined feature request. Perhaps you could look around through the github issues and see if it has been reported already and, if not, report a new issue just for that? It makes it easier to keep track of things if there's just one bug or feature per github issue. Even better would be if you could implement it and submit a pullrequest for it :-) Same for the color preferences, I think there might already be an issue for that.

ffissore, 2015-05-29:
IDE reopens sketches when closed with Quit. Fixed with  cc4af81ac54313da03b0733cc754eb1f3d040374 and available with latest hourly build http://www.arduino.cc/en/Main/Software#hourly
FYI, "File > Recent" menu is under development

closed 2015-05-29 by ffissore