arduino/Arduino #2558: SPI pin connection on arduino prototype shield is wrong

rybus, 2015-01-18:
Hi,

On the documentation http://arduino.cc/en/Main/ArduinoProtoShield there is SPI pin mapping. Which is right only if using Arduino Uno. If using Arduino Yun, only rest, +5V and GND are well connected.

All of the others are not. And can't be used as they not connected but soldered in the connector... This should corrected in the documentation and explained the special arduino yun case.

I think these pins should really be connected as explained in the doc.

agdl, 2015-07-10:
@rybus i corrected it with "On the ICSP connector only 5V, GND and RST are wired tot the respective pins on the header. MOSI and MISO are present only on the connector." which is the truth. If it is fine for you please close this issue.

rybus, 2015-07-10:
Okay perfect :)

closed 2015-07-10 by rybus