arduino/Arduino #2560: 9x performance degradation in 1.5.7, 1.5.8 1.6.0 against 1.5.6-2 and 1.0.6

BertHav, 2015-01-19:
When building the attached sketch in 1.5.7 1.5.8 and the nightly build the sketch fails to execute or it takes about 60 seconds to complete. When build in 1.5.6-2 or 1.0.6 it works like a charm and completes in about 6500ms. Used hardware is UNO R3 with 5100 R6 shield and Mega2560 R3 and the same shield. The observed behaviour is the same for both an UNO and the Mega configuration. 
It looks to me that the ethernetstream times out and wait for a retransmit after about 5 seconds when reading data. The data comes in in chunks of about 3.5kB. The total received datastream is about 47kB.
When compiled in 1.0.6 or 1.5.6-2 it is an uninterrupted datastream.

The attached sketch is part of reading an energy meter, but stripped to reproduce the problem only.
Do I overlook something?
Thanks in advance,
Bert

// Retrieves the actual weatherinformation from xml.buienradar.nl

#include <SPI.h>
#include <Ethernet.h>
#include <EthernetClient.h>
#include <Dns.h>
#include "Time.h"
#include <FlashMini.h>

float actual;
char weatherType[25];
float totPluvio;
float pluvio;
int lv_phi;
float windsnelheidMS;
char windrichting[4];
float luchtdruk;
unsigned int visibility;
char icon[64];
char samenvatting[256];
char meetStation[25];

float MinTemp;                                // Minimum Temperature
float MaxTemp;                                // Maximum Temperature
float prevActual;  // the previous temperature from the metering meteo station
unsigned int todayCnt;
unsigned int todayErrCnt;
//  byte Type;                                   // Variable of PvOutput to log to. See userdefs.h for explanation
char TimeMinTemp[6];
char TimeMaxTemp[6];
time_t t_tijd;
//start including ethernet
// Enter a MAC address and IP address for your controller below.
byte mac[] = {
  0x08, 0x00, 0x2B, 0xB1, 0xFE, 0xED
}; //My DEC history ;)
IPAddress ip(192, 168, 1, 3);
// in this sketch the iptx will be overwritten by the Rocrail server
// address during init. This sketch detects the server address automatic
// if this device is configured as a MCS2 controller with our ip address
IPAddress iptx(192, 168, 1, 255); //Address for Rocrail

// the router's gateway address:
byte gateway[] = { 192, 168, 1, 254 };
// the subnet:
byte subnet[] = { 255, 255, 255, 0 };
static byte dnsserver[] = {192, 168, 1, 254};

void setup()
{
  sprintf(TimeMinTemp, "%02d:%02d", 0, 0);
  sprintf(TimeMaxTemp, "%02d:%02d", 0, 0);
  Serial.begin( 115200 );      // console terminal for log info
  Serial.print(F("MAC: "));
  char webData[32];
  sprintf(webData, "%02X-%02X-%02X-%02X-%02X-%02X", mac[0], mac[1], mac[2], mac[3], mac[4], mac[5]);
  Serial.println(webData);
  Serial.println();
  // start the Ethernet and UDP:
  // Sprinkle some magic pixie dust. (disable SD SPI on the Ethernetshield)
  pinMode(4, OUTPUT);
[... 164 lines cut ...]
          Stringbuff[254] = '\0';
        }
        Stringbuff.toCharArray(samenvatting, 255);
      }
      unsigned long p_tijd = millis() - startWeatherProc;
      Serial.print(F("weather processing time in ms: "));
      Serial.println(p_tijd);
      Serial.println(F("Weather updated"));
    }
    else {
      Serial.println(F("weatherStation not found"));
    }
    buienradarClient.stop();
  }
  else {
    Serial.println(F("No connection with xml.buienradar.nl"));
    todayErrCnt++;
  }
}

void loop()
{
  GetTemperature();
  delay(1000);
}

ffissore, 2015-01-22:
Can you also provide the libraries you're using? They are FlashMini and Time

BertHav, 2015-01-22:
Hi Federico,

 

Attached the libraries I use.

 

Regards,

 

Bert

ffissore, 2015-01-22:
Wherever: dropbox, google drive...

BertHav, 2015-01-22:
https://drive.google.com/folderview?id=0B_fITTqKEGXAa3l2Q2ZEeC1Mdlk&usp=sharing

ffissore, 2015-09-17:
On my computer latest nightly takes 3.95 seconds to compile your sketch. On windows, antivirus software may significantly slow down things

closed 2015-09-17 by ffissore