arduino/Arduino #2562: JAVA bug with MAC OS X

jlefevre2, 2015-01-20:
Arduino has a bug common to several other programs that work on Mac OS X.  If you have a clean install of Mac OS X it will not have JAVA. You can go to Oracles site and download JAVA 7 or 8 and it will install properly on the Mac. Most JAVA programs will start and correctly find JAVA and run okay. There are a few , However, that will not find a properly installed JAVA ( this is the case with Arduino ) and will only look where Apple has installed JAVA 6 ( now obsolete and unsupported and soon to not be on Apples download page). Since Arduino looks in the Apple location to determine if JAVA is present ( which is a wrong and nonstandard location BTW) it pops up the message " You must install JAVA 6 from APPLE " even though there is a proper install of JAVA 7 or 8 from ORACLE on the Mac and that install is proper and working. I had the same problem with the JMRI program and they finally figured out the problem and put the correct path name in JMRI so it would look in the standard Oracle install ( not the Apple JAVA 6 install ) and then JMRI would work. In any case the only work around is to install Apple JAVA 6 in addition to JAVA 8 from oracle and Arduino will then start and run under JAVA 8. This needs fixed very soon as JAVA 6 will be leaving the Mac web site pretty soon. This applies to all version of the MAC OS X operating system that some without JAVA. So that is 10 and 9 and 8 and 7 for sure. If this is not fixed the Arduino will cease to work on Mac's as soon and Apple JAVA 6 goes away.

cmaglie, 2015-01-27:
@jlefevre2 
Is this happening also with the IDE 1.5.8 for Java 7?
You can try it here, in case: http://arduino.cc/download.php?f=/arduino-1.5.8-macosx-java7.zip

jlefevre2, 2015-01-27:
Yes. Won't start.

ffissore, 2015-02-01:
@jlefevre2 can you try version "macosx-new" from https://github.com/arduino/Arduino/pull/2590#issuecomment-72337128 ?

jlefevre2, 2015-02-01:
> @jlefevre2 <https://github.com/jlefevre2> can you try version "macosx-new" from #2590 (comment) <https://github.com/arduino/Arduino/pull/2590#issuecomment-72337128> ?

Tried it. Works fine. Took me a few minutes to find a Mac without APPLE java ever installed and ONLY Oracle java. 

Looks like a winner. 

Thanks

ffissore, 2015-02-11:
Fixed with IDE 1.6.0 release

closed 2015-02-11 by ffissore