arduino/Arduino #2564: Wire Library, Explored documentation clarity

skippyV, 2015-01-21:
At the end of this doc:
http://playground.arduino.cc/Main/WireLibraryDetailedReference
Is this text:
"It must call twi_transmit() in order to fill the transmit buffer with data. It also must call the Wire library's send() method to fill the transmit buffer; see send()."
After reading this I'm even more confused. It implies there are 2 transmit buffers?

stickbreaker, 2015-06-15:
Out of date documentation.

Only 1 buffer.
```c++
onRequestEvent(){
// to return 3 bytes
   Wire.write(1);
   Wire.write(2);
   Wire.write(3);
// or
uint8_t b[3]={1,2,3};
  Wire.write(b,3);
}
```

Chuck

cmaglie, 2015-07-01:
The text is on the playground and it's editable by anyone, feel free to fix it.

closed 2015-07-01 by cmaglie