arduino/Arduino #2571: Ethernet Library: DHCP Spam

Elijahg, 2015-01-23:
This has been reported in the forum but I'd like to 'officially' report it here since it's seemingly gone ignored for 3 years, and the problem still exists.

When using Ethernet.maintain(), there are a few bugs that cause the Arduino to constantly spam the network with DHCP broadcast requests - not great for battery powered devices like phones that have to awaken to process all this data (around 50k/sec I was getting).

One bug was reported on the old tracker as issue #985. That's not fixed yet, but the new one was on the forum here: http://forum.arduino.cc/index.php?topic=195422.0 . Essentially half the DHCP reply is missing, and the Library relies on the DHCP server _not_ following the RFC and thus dishing out an existing address. DHCP servers that do follow the RFC hand out a new address, confusing things.

Both are fairly important as not to break people's networks, and fixes have been provided. It'd be a 2 minute job to fix.

I've checked the source for 1.5.x and it's not fixed there, nor in the live 1.0.6 release.

chaveiro, 2015-02-15:
See my comments on this before merging : https://github.com/arduino/Arduino/commit/bde3d715bd25a2ce16f84c43d93e4acb77c16f23#comments

cmaglie, 2015-06-04:
A proposed fix is in #3287 please follow up there.

commit a33c93b1 (2015-07-14): Ethernet: fixed wrong handling of timeouts in DHCP

The signed math doesn't handle correctly cases where the lease
time is set to infinity (0xFFFFFFFF).

Fixes #2571
Fixes #2601
Fixes #2642
Fixes #985
closed 2015-06-04 by cmaglie