arduino/Arduino #2572: Arduino Nano Eagle files archive is corrupt

DarienSJ, 2015-01-23:
The Arduino Nano schematic & design Eagle files archive is not extractable & Ark lists no contents.
http://arduino.cc/en/uploads/Main/ArduinoNano30Eagle.zip is not extractable & shows no contents.
Can it please be replaced with a good one?

'''
 18:27:09 Fri 2015-01-23 user@Sanitized Using bash on 2 with 0 jobs 
 31\517 /EtA/#####Data Files/Electronics/Eagle/Arduino: $ wget http://arduino.cc/en/uploads/Main/ArduinoNano30Eagle.zip
--2015-01-23 18:27:12--  http://arduino.cc/en/uploads/Main/ArduinoNano30Eagle.zip
Resolving arduino.cc... 174.129.243.245
Connecting to arduino.cc|174.129.243.245|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 770048 (752K) [application/zip]
Saving to: ‘ArduinoNano30Eagle.zip’

ArduinoNano30Eagle.zip         100%[===================================================>] 752.00K   204KB/s   in 3.9s   

2015-01-23 18:27:16 (192 KB/s) - ‘ArduinoNano30Eagle.zip’ saved [770048/770048]

 18:32:21 Fri 2015-01-23 user@Sanitized Using bash on 2 with 0 jobs 
 41\527 /EtA/#####Data Files/Electronics/Eagle/Arduino: $ unzip -l ./ArduinoNano30Eagle.zip 
Archive:  ./ArduinoNano30Eagle.zip
  End-of-central-directory signature not found.  Either this file is not
  a zipfile, or it constitutes one disk of a multi-part archive.  In the
  latter case the central directory and zipfile comment will be found on
  the last disk(s) of this archive.
note:  ./ArduinoNano30Eagle.zip may be a plain executable, not an archive
unzip:  cannot find zipfile directory in one of ./ArduinoNano30Eagle.zip or
        ./ArduinoNano30Eagle.zip.zip, and cannot find ./ArduinoNano30Eagle.zip.ZIP, period.

 18:32:30 Fri 2015-01-23 user@Sanitized Using bash on 2 with 0 jobs 
 42\528 /EtA/#####Data Files/Electronics/Eagle/Arduino: $ unzip -t ./ArduinoNano30Eagle.zip 
Archive:  ./ArduinoNano30Eagle.zip
  End-of-central-directory signature not found.  Either this file is not
  a zipfile, or it constitutes one disk of a multi-part archive.  In the
  latter case the central directory and zipfile comment will be found on
  the last disk(s) of this archive.
note:  ./ArduinoNano30Eagle.zip may be a plain executable, not an archive
unzip:  cannot find zipfile directory in one of ./ArduinoNano30Eagle.zip or
        ./ArduinoNano30Eagle.zip.zip, and cannot find ./ArduinoNano30Eagle.zip.ZIP, period.

 18:27:16 Fri 2015-01-23 user@Sanitized Using bash on 2 with 0 jobs 
 32\518 /EtA/#####Data Files/Electronics/Eagle/Arduino: $ unzip ./ArduinoNano30Eagle.zip 
Archive:  ./ArduinoNano30Eagle.zip
  End-of-central-directory signature not found.  Either this file is not
  a zipfile, or it constitutes one disk of a multi-part archive.  In the
  latter case the central directory and zipfile comment will be found on
  the last disk(s) of this archive.
note:  ./ArduinoNano30Eagle.zip may be a plain executable, not an archive
unzip:  cannot find zipfile directory in one of ./ArduinoNano30Eagle.zip or
        ./ArduinoNano30Eagle.zip.zip, and cannot find ./ArduinoNano30Eagle.zip.ZIP, period.

 18:27:23 Fri 2015-01-23 user@Sanitized Using bash on 2 with 0 jobs 
 33\519 /EtA/#####Data Files/Electronics/Eagle/Arduino: $ type ./ArduinoNano30Eagle.zip 
./ArduinoNano30Eagle.zip is ./ArduinoNano30Eagle.zip

 18:27:54 Fri 2015-01-23 user@Sanitized Using bash on 2 with 0 jobs 
 34\520 /EtA/#####Data Files/Electronics/Eagle/Arduino: $ file ./ArduinoNano30Eagle.zip 
./ArduinoNano30Eagle.zip: Zip archive data, at least v2.0 to extract

 18:27:58 Fri 2015-01-23 user@Sanitized Using bash on 2 with 0 jobs 
 35\521 /EtA/#####Data Files/Electronics/Eagle/Arduino: $ mv ./ArduinoNano30Eagle.zip ./ArduinoNano30Eagle

 18:28:06 Fri 2015-01-23 user@Sanitized Using bash on 2 with 0 jobs 
 36\522 /EtA/#####Data Files/Electronics/Eagle/Arduino: $ file ./ArduinoNano30Eagle 
./ArduinoNano30Eagle: Zip archive data, at least v2.0 to extract

 18:28:23 Fri 2015-01-23 user@Sanitized Using bash on 2 with 0 jobs 
 37\523 /EtA/#####Data Files/Electronics/Eagle/Arduino: $ unzip ./ArduinoNano30Eagle 
Archive:  ./ArduinoNano30Eagle
  End-of-central-directory signature not found.  Either this file is not
  a zipfile, or it constitutes one disk of a multi-part archive.  In the
  latter case the central directory and zipfile comment will be found on
  the last disk(s) of this archive.
note:  ./ArduinoNano30Eagle may be a plain executable, not an archive
unzip:  cannot find zipfile directory in one of ./ArduinoNano30Eagle or
        ./ArduinoNano30Eagle.zip, and cannot find ./ArduinoNano30Eagle.ZIP, period.

 18:28:29 Fri 2015-01-23 user@Sanitized Using bash on 2 with 0 jobs 
 38\524 /EtA/#####Data Files/Electronics/Eagle/Arduino: $ rm ./ArduinoNano30Eagle 

 18:28:32 Fri 2015-01-23 user@Sanitized Using bash on 2 with 0 jobs 
 39\525 /EtA/#####Data Files/Electronics/Eagle/Arduino: $ 
'''

DarienSJ, 2015-01-23:
'''
 18:27:09 Fri 2015-01-23 user@Sanitized Using bash on 2 with 0 jobs 
 31\517 /EtA/#####Data Files/Electronics/Eagle/Arduino: $ wget http://arduino.cc/en/uploads/Main/ArduinoNano30Eagle.zip
--2015-01-23 18:27:12--  http://arduino.cc/en/uploads/Main/ArduinoNano30Eagle.zip
Resolving arduino.cc... 174.129.243.245
Connecting to arduino.cc|174.129.243.245|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 770048 (752K) [application/zip]
Saving to: ‘ArduinoNano30Eagle.zip’

ArduinoNano30Eagle.zip         100%[===================================================>] 752.00K   204KB/s   in 3.9s   

2015-01-23 18:27:16 (192 KB/s) - ‘ArduinoNano30Eagle.zip’ saved [770048/770048]

 18:32:21 Fri 2015-01-23 user@Sanitized Using bash on 2 with 0 jobs 
 41\527 /EtA/#####Data Files/Electronics/Eagle/Arduino: $ unzip -l ./ArduinoNano30Eagle.zip 
Archive:  ./ArduinoNano30Eagle.zip
  End-of-central-directory signature not found.  Either this file is not
  a zipfile, or it constitutes one disk of a multi-part archive.  In the
  latter case the central directory and zipfile comment will be found on
  the last disk(s) of this archive.
note:  ./ArduinoNano30Eagle.zip may be a plain executable, not an archive
unzip:  cannot find zipfile directory in one of ./ArduinoNano30Eagle.zip or
        ./ArduinoNano30Eagle.zip.zip, and cannot find ./ArduinoNano30Eagle.zip.ZIP, period.

 18:32:30 Fri 2015-01-23 user@Sanitized Using bash on 2 with 0 jobs 
 42\528 /EtA/#####Data Files/Electronics/Eagle/Arduino: $ unzip -t ./ArduinoNano30Eagle.zip 
Archive:  ./ArduinoNano30Eagle.zip
  End-of-central-directory signature not found.  Either this file is not
  a zipfile, or it constitutes one disk of a multi-part archive.  In the
  latter case the central directory and zipfile comment will be found on
  the last disk(s) of this archive.
note:  ./ArduinoNano30Eagle.zip may be a plain executable, not an archive
unzip:  cannot find zipfile directory in one of ./ArduinoNano30Eagle.zip or
        ./ArduinoNano30Eagle.zip.zip, and cannot find ./ArduinoNano30Eagle.zip.ZIP, period.

 18:27:16 Fri 2015-01-23 user@Sanitized Using bash on 2 with 0 jobs 
 32\518 /EtA/#####Data Files/Electronics/Eagle/Arduino: $ unzip ./ArduinoNano30Eagle.zip 
Archive:  ./ArduinoNano30Eagle.zip
  End-of-central-directory signature not found.  Either this file is not
  a zipfile, or it constitutes one disk of a multi-part archive.  In the
  latter case the central directory and zipfile comment will be found on
  the last disk(s) of this archive.
note:  ./ArduinoNano30Eagle.zip may be a plain executable, not an archive
unzip:  cannot find zipfile directory in one of ./ArduinoNano30Eagle.zip or
        ./ArduinoNano30Eagle.zip.zip, and cannot find ./ArduinoNano30Eagle.zip.ZIP, period.

 18:27:23 Fri 2015-01-23 user@Sanitized Using bash on 2 with 0 jobs 
 33\519 /EtA/#####Data Files/Electronics/Eagle/Arduino: $ type ./ArduinoNano30Eagle.zip 
./ArduinoNano30Eagle.zip is ./ArduinoNano30Eagle.zip

 18:27:54 Fri 2015-01-23 user@Sanitized Using bash on 2 with 0 jobs 
 34\520 /EtA/#####Data Files/Electronics/Eagle/Arduino: $ file ./ArduinoNano30Eagle.zip 
./ArduinoNano30Eagle.zip: Zip archive data, at least v2.0 to extract

 18:27:58 Fri 2015-01-23 user@Sanitized Using bash on 2 with 0 jobs 
 35\521 /EtA/#####Data Files/Electronics/Eagle/Arduino: $ mv ./ArduinoNano30Eagle.zip ./ArduinoNano30Eagle

 18:28:06 Fri 2015-01-23 user@Sanitized Using bash on 2 with 0 jobs 
 36\522 /EtA/#####Data Files/Electronics/Eagle/Arduino: $ file ./ArduinoNano30Eagle 
./ArduinoNano30Eagle: Zip archive data, at least v2.0 to extract

 18:28:23 Fri 2015-01-23 user@Sanitized Using bash on 2 with 0 jobs 
 37\523 /EtA/#####Data Files/Electronics/Eagle/Arduino: $ unzip ./ArduinoNano30Eagle 
Archive:  ./ArduinoNano30Eagle
  End-of-central-directory signature not found.  Either this file is not
  a zipfile, or it constitutes one disk of a multi-part archive.  In the
  latter case the central directory and zipfile comment will be found on
  the last disk(s) of this archive.
note:  ./ArduinoNano30Eagle may be a plain executable, not an archive
unzip:  cannot find zipfile directory in one of ./ArduinoNano30Eagle or
        ./ArduinoNano30Eagle.zip, and cannot find ./ArduinoNano30Eagle.ZIP, period.

 18:28:29 Fri 2015-01-23 user@Sanitized Using bash on 2 with 0 jobs 
 38\524 /EtA/#####Data Files/Electronics/Eagle/Arduino: $ rm ./ArduinoNano30Eagle 

 18:28:32 Fri 2015-01-23 user@Sanitized Using bash on 2 with 0 jobs 
 39\525 /EtA/#####Data Files/Electronics/Eagle/Arduino: $ 
'''

yozik04, 2015-01-28:
Same thing for me. This zip file is corrupt.

amunter-nist, 2015-01-30:
Same here.  Trying to download Eagle schematic for Nano, but zip is corrupt.  An older 3.0 download from November also had the Atmega168 in it, which I think is the 2.3 version, not the 3.0.

maxlazarus, 2015-02-13:
Would like to see this looked at as well.

haganwalker, 2015-02-27:
This is definitely a corrupt zip and I can't seem to find these files elsewhere.

beniroquai, 2015-03-02:
Yep. corrupt!

agdl, 2015-04-03:
Hi to all! I'm able to open it and it works. I used eagle 7.1. Maybe someone re-uploaded it according to this issue without closing this. If it doesn't work please reopen this.

closed 2015-04-03 by agdl