arduino/Arduino #2574: Hitting compile twice causes error, have to restart IDE

NicoHood, 2015-01-24:
Just hit the compile button when you are already compiling. The error wont disappear, you have to restart the IDE.

```
C:\Users\Nico\AppData\Local\Temp\build4870134429926225134.tmp/core.a(main.cpp.o): In function `main':
C:\Users\Nico\Dropbox\Arduino\sketchbook\hardware\HID\avr\cores\hid/main.cpp:32: undefined reference to `init'
C:\Users\Nico\AppData\Local\Temp\build4870134429926225134.tmp/core.a(USBCore.cpp.o): In function `USB_ClockEnable':
C:\Users\Nico\Dropbox\Arduino\sketchbook\hardware\HID\avr\cores\hid\USB-Core/USBCore.cpp:742: undefined reference to `delay'
C:\Users\Nico\AppData\Local\Temp\build4870134429926225134.tmp/core.a(USBCore.cpp.o): In function `USB_Send(unsigned char, void const*, int)':
C:\Users\Nico\Dropbox\Arduino\sketchbook\hardware\HID\avr\cores\hid\USB-Core/USBCore.cpp:316: undefined reference to `delay'
collect2: error: ld returned 1 exit status
Error compiling.
C:\Users\Nico\AppData\Local\Temp\build4870134429926225134.tmp/core.a(main.cpp.o): In function `main':
C:\Users\Nico\Dropbox\Arduino\sketchbook\hardware\HID\avr\cores\hid/main.cpp:32: undefined reference to `init'
C:\Users\Nico\AppData\Local\Temp\build4870134429926225134.tmp/core.a(USBCore.cpp.o): In function `USB_ClockEnable':
C:\Users\Nico\Dropbox\Arduino\sketchbook\hardware\HID\avr\cores\hid\USB-Core/USBCore.cpp:742: undefined reference to `delay'
C:\Users\Nico\AppData\Local\Temp\build4870134429926225134.tmp/core.a(USBCore.cpp.o): In function `USB_Send(unsigned char, void const*, int)':
C:\Users\Nico\Dropbox\Arduino\sketchbook\hardware\HID\avr\cores\hid\USB-Core/USBCore.cpp:316: undefined reference to `delay'
collect2: error: ld returned 1 exit status
Error compiling.
```

Using IDE 1.6 rc2 Windows 8.1 Update

ffissore, 2015-01-26:
Just tried with a windows 7: the error is shown but disappears when I hit compile once more after the first two compilations terminate

NicoHood, 2015-01-28:
Hm looks the same for me now with uno BUT not for leonardo boards. Try that pls.

NicoHood, 2015-02-12:
anyone?

ffissore, 2015-09-08:
Merged into #3771

closed 2015-09-08 by ffissore