arduino/Arduino #2575: Unplugging Uno with IDE open causes kernel panic (OSX Yosimite)

hestonkent, 2015-01-25:
Unfortunately I can't find the error codes to provide on this within the console, and no shutdown dialog box appears upon reboot with an error report. However there's documentation all over the web with this occurrence, that can be found here: http://tinyurl.com/ooq8oj9
 

This occurs with sketches that even have no serial communication once uploaded whatsoever. Extremely frustrating, and discouraging as someone who is trying to learn Arduino.

bobc, 2015-01-25:
Duplicate of #350? Try version 1.5.6 or later.

Otherwise, seems to be a bug in OSX, needs Apple to fix it.

ffissore, 2015-01-25:
@hestonkent what version of the IDE are yo uusing?

hestonkent, 2015-01-26:
Ah! I was using the 1.0.5, which appeared to be the 'non-beta' release. After upgrading to the 1.5.8, I cannot replicate the problem. Whoop whoop!

closed 2015-01-26 by hestonkent