arduino/Arduino #2576: cannot find YUN  in 1.6.0

jl-p, 2015-01-25:
Hi,

I have a Yun connected to my network and running 1.5.8 and 1.6 on the same laptop
when i start the 1.5.8 version it detects my yun. 
when i start the 1.6.0 version it won't 

Johan

TrippyLighting, 2015-01-25:
Did you mean to say 1.0.6 ?
That IDE does not support the Yún. The Due and the Yun are t main reasons behind the 1.5.x IDE

ffissore, 2015-01-26:
What operating system are you using?

jl-p, 2015-01-26:
windows 8.
And no 1.6.0 nightly build.

ffissore, 2015-01-26:
nightly builds are not signed. Could this be the source of the problem?

jl-p, 2015-01-26:
I Know.
But if no one reports it. it will be in the final version also.

ffissore, 2015-01-26:
If the source of the problem is the missing signature, then no: the final release will be signed

NicoHood, 2015-01-28:
I played with my yun shield a few days ago and noticed that the board appeared and disappeard sometimes very bad. Not sure if this is a 1.6 bug since ive never tested it with 1.5.8. But I also noticed some problems with the connection.

ffissore, 2015-02-11:
@jl-p 1.6.0 has been released. Have you already tried it?

roadfun, 2015-02-16:
I was also having several problems with Yun discovery (I have 3 Yuns and my issues started prior to 1.6). I did some debugging in the IDE and found some problematic code. I submitted a code change that has resolved these issues for me and now the Discovery is very reliable. Hopefully the fix is a good one and will then help others. I've also seen posts about this problem on the Arduino Yun forum.

agdl, 2015-03-13:
@jl-p & @ffissore reopen it if needed

closed 2015-03-13 by agdl

ffissore, 2015-03-13:
Indeed fixed (or at least mitigated) in 1.6.1