arduino/Arduino #2577: Arduino Starter Kit Materials

knildon, 2015-01-25:
I recently purchased a starter kit and have enjoyed working on the projects and learning the coding system.  The only drawbacks I have found so far are the jumper wires and the male header pins.
Each project requires at least one or more sets of short power (red and black) jumpers for use on the breadboard.  My kit does not have any black jumpers and only long red jumpers.  It does have some very short red jumpers but they are to short for use as a power feed on the board.
The male header pins have a short and a long side to them.  Unfortunately the short side is too short to use on the breadboard and also too short to use on the servo connector as shown in the instructions.
I believe these two problems should be addressed for future kits.  I worked around them but it would be nice to have what is shown in the projects book.
Don

ffissore, 2015-01-30:
Hi Don and thanks for your suggestion
Can you please mail support@arduino.cc and provide information about your starter kit: a picture of it, where you bought it and the invoice if you have one

knildon, 2015-01-31:
Unfortunately my UNO stopped working.  It kept giving me some sort of mismatch error.  I have since returned it.  I found a couple of other problems before it quit working but did not document everything.  One I remember was the tilt switch was not the same as in the book.  Mine was 4 pins and they were to short to plug into the breadboard.  No pics available but I can contact support if you think it will help.  I'm a little upset about the UNO because I couldn't find anywhere what the error was and what caused it.  I will probably buy another kit when I get my refund since I enjoy doing that sort of work.
Don

ffissore, 2015-01-31:
It looks weird for an official Arduino Starter Kit to have such issues. Closing this issue: it will eventually be handled by support team, once contacted

closed 2015-01-31 by ffissore