arduino/Arduino #2579: Document F() macro for placing constant strings in program (flash) memory.

damellis, 2015-01-26:
You can use the F() macro to place constants in flash memory, not RAM, e.g. Serial.println(F("foo")) but I didn't find this documented anywhere in the Arduino reference.

matthijskooijman, 2015-01-26:
The use of the __FlashStringHelper* type for storing such constants should probably also be documented.

On a related note, the http://arduino.cc/en/Reference/PROGMEM should also be updated. Since the gcc and libc upgrade in 1.5.7 (IIRC), there is no need to use the prog_* types anymore.

q2dg, 2015-04-06:
It is documented in http://playground.arduino.cc/Learning/Memory 

With regard to PROGMEM, I think http://arduino.cc/en/Reference/PROGMEM is updated (see #2647), but http://playground.arduino.cc/Main/PROGMEM is not...who cares? 

This issue should be closed

agdl, 2015-09-28:
@q2dg @damellis documentation added for Serial.print(F()) added https://www.arduino.cc/en/Reference/PROGMEM

@matthijskooijman how do you suggest to modify the current documentation?

q2dg, 2015-09-28:
* There's some typo ("istruction" , the comma between "to be printed" and "is normally"...)
* Maybe behind "PROGMEM is part of the pgmspace.h library" sentence I would specify ("which is only available in AVR architecture") 
* The paragraph talking about F() is placed in a nonsense way in a nonsense place. It should be better written (with a proper title, like 'The F() macro' or similar) at the end of all text.

agdl, 2015-09-29:
@q2dg modified in the way you suggested

q2dg, 2015-09-29:
Ok, thanks. For me it's already right

agdl, 2015-09-29:
ok, @matthijskooijman can this be closed?

agdl, 2015-10-07:
I'm closing this, reopen it if necessary

closed 2015-10-07 by agdl