arduino/Arduino #2580: Rename / alias __FlashStringHelper (e.g. to "FlashString").

damellis, 2015-01-26:
We have a "friendly" alias for creating flash strings (the F() macro), but not for using them within a library. This is relevant for anyone who wants to create something that sub-classes Print (or Stream) and can print these flash-resident strings. For those people, it seems like it would be nice to have a nicer name than __FlashStringHelper (e.g. FlashString) and to document this somewhere.

fingolfin, 2016-01-04:
+1 to this

PaulStoffregen, 2016-01-04:
Will all this stuff become unnecessary when Arduino upgrades to a version of avr-gcc with named address spaces?

m-ri, 2016-09-11:
Ciao,

Ho alcune informazioni che  possono essere molto utili per te, dai un’occhiata <http://elisa.scamsli.org/gjj>

Sinceramente,marco ribero