arduino/Arduino #2583: Linux arduino shell script will not open INO filename passed as parameter

VikOlliver, 2015-01-27:
Invoking the arduino shell script on a Linux system in the form "arduino mycode.ino" will not open the mycode.ino file.

This is because the arduino shell script changes to the arduino executable directory before passing the filename parameter to the java executable. The filename supplied by the user is then no longer in the current directory.

This can be fixed by using the readlink command to acquire the fully qualified filename before changing directory, then passing this to the java app. This does not require any resources not used in the original arduino shell script. Suggested fix:

#!/bin/sh
 
if [ "x" != "x$@" ] ; then
    FULLFILE=$(readlink -f "$@")
fi
APPDIR="$(dirname -- "$(readlink -f -- "${0}")" )"

cd "$APPDIR"
 
for LIB in \
    java/lib/rt.jar \
    java/lib/tools.jar \
    lib/*.jar \
    ;
do
    CLASSPATH="${CLASSPATH}:${LIB}"
done
export CLASSPATH

LD_LIBRARY_PATH=`pwd`/lib${LD_LIBRARY_PATH:+:$LD_LIBRARY_PATH}
export LD_LIBRARY_PATH

export PATH="${APPDIR}/java/bin:${PATH}"
java -Dswing.defaultlaf=com.sun.java.swing.plaf.gtk.GTKLookAndFeel processing.app.Base "$FULLFILE"

matthijskooijman, 2015-01-27:
I actually think this is fixed in the most recent 1.5.x versions by passing a -workdir or something like that so the arduino command can resolve relative links correctly (which only works correctly on Linux currently IIRC). What version are you using?

VikOlliver, 2015-01-27:
Version 1.0.6 32-bit, which I downloaded yesterday. I'm running on Ubuntu.

matthijskooijman, 2015-01-28:
Right. 1.0.x doesn't really support any commandline options at all (though apparently it does support passing a .ino file, didn't know that). In any case, this will be fixed in the upcoming stable release 1.6.0 (there is already a release candidate available that you could test). I'm closing this ticket, if you think this is wrong, please follow up.

closed 2015-01-28 by matthijskooijman