arduino/Arduino #2584: Faulty function invocation fails without error message.

RussellMcMahon, 2015-01-27:
Trying to invoke a function 
            SomeName();
using 
            SomeName;  //              <- omitted ()

Does not produce an error message on compiling.
And, of course, it also does not call AllOff()

Notes:

1. There are no other variables, constants, functions etc with AllOff anywhere in their name.
2. The problem is in some way related to the valid AllOff() 
    As
                 AllOff;   // <- no error message, no function call. 
   But eg
                RandomName;  // <- error message produced.  
Gives:
    BT_V0504.ino: In function 'void MainLoop1()':
    BT_V0504:215: error: 'RandomName' was not declared in this scope

3. Arduino version 1.0.6
    Windows 8.0

4. Function AllOff() declared by:

void AllOff()
{
    misc code ; 
    ....             ;
}

Other functions produce the same (lack of) result when the () are omitted.

ffissore, 2015-01-30:
Can you share the whole sketch?

agdl, 2015-04-09:
reopen this if needed

closed 2015-04-09 by agdl