arduino/Arduino #2596: Readme refers to non-existing instructions

probonopd, 2015-02-02:
The README.md on https://github.com/arduino/Arduino/ refers to  `reference/Guide_Windows.html` and `reference/Guide_MacOSX.html`but these do not appear to exist.

probonopd, 2015-02-02:
I am looking for instructions on how to compile the IDE from this repository.

ffissore, 2015-02-03:
Those files are in the reference.zip file, that gets extracted when the IDE is packaged.
If you want to build the IDE, `cd` into folder `build` and run `ant clean build run`

probonopd, 2015-02-07:
Thanks. Looks like `ant clean build ; echo "" | ant dist` is what I was looking for. It creates a binary build from the source code. Maybe this could be posted on the README.md of this repo under a headline like "Building"?

ffissore, 2015-02-07:
A shorter syntax is `ant clean dist`. Since you've just started hacking with the IDE, what about changing the Readme yourself and make a pull request? :)

ffissore, 2015-02-11:
Build instructions will be ported and updated with #2621. Closing

closed 2015-02-11 by ffissore