arduino/Arduino #2598: Arduino UNO not recognized on 20 brand new boards and 20 different PCs

pzenzerovic, 2015-02-02:
Hy,

I am the distributor for Arduino for Croatia. I just got my new shipment of 30 Arduino UNO boards. I came to hold a workshop with them and gave them to the users (about 20 users with 20 laptops from WIN XP to Win 8). They all had installed 1.0.6 IDE and drivers. When they plugged the boards not a single one was recognized as an Arduino UNO. All of the devices on all of the computers were recognized as Unknown device. Then I gave them 15 Arduino UNOs from my previous shipment and everything worked OK without any changes to the laptops. I have never seen a problem like this, even though I have installed the Arduino software and drivers on a few thousand PCs while holding lectures.

As I have a Macbook I took one of the Arduino UNO boards that was not recognized on the mentioned laptops and plugged it in my Mac. It was instantly recognized and worked without any problems. 

When I placed the same board back on a few of the mentioned laptops it still wasnt recognized.

After the workshop I decided to try it out on my Windows laptop and the same thing happend again. The board was not recognized. I wnt to device manager and updated my driver manually by choosing "let me pick from i list of drivers..." and choose Arduino UNO from the Arduino LLC and it installed correctly and works just fine. But when I plug a different UNO on the same laptop the problem persists.

Can anyone point me to the problem?

Thanks
Paolo Zenzerovic
Arduino distributor for Croatia

ffissore, 2015-02-03:
Hi Paolo, please forward your message to support@arduino.cc. My colleagues will contact you right after

pzenzerovic, 2015-02-03:
Thanks, I have forwarded the message.

closed 2015-02-03 by ffissore