arduino/Arduino #2600: Extra Due Serial ports, like Mega

sayf19, 2015-02-03:
The documentation correctly describes the extra serial ports on the Mega board, Serial, Serial1, Serial2, Serial3.
However, it makes no mention of the Due (SAM3X), where these ports are also available.  You could just add a word or 2 to the Serial library pages that describe this to say that this works for the Mega AND the Due.

Hope this helps someone!

ph1ash, 2015-02-25:
+1 on this. Serial1 isn't working on my Due, not sure why. Going through the libraries and noticed that HardwareSerial1,2,3 aren't in the "sam" folder but they're within the "avr" folder. Might need to do some porting...

cmaglie, 2015-02-25:
@ph1ash 
AFAIK the Serial1/2/3 works on the Due, and the OP is just asking to document that in the reference guide.

If Serial1 is not working for you, you should probably ask in the forum for help to sort out your problem.

agdl, 2015-04-03:
@sayf19 here

http://arduino.cc/en/Reference/Serial

 the ports are described. Can you link the page you are referring to?

agdl, 2015-04-09:
Reopen the issue if needed

closed 2015-04-09 by agdl

sayf19, 2015-04-09:
Looks like some explanation was added regarding the Due and the extra 
serial ports.  It looks correct now.

Thanks!