arduino/Arduino #2603: error while compiling my program to arduino yun

karthisl, 2015-02-06:
my program
//
#include "pulse-sensor-arduino.h"
#include "Arduino.h"
#define PulseSensor::begin()
//  VARIABLES
int pulsePin = 0;
int blinkPin = 13;                // pin to blink led at each beat
int fadePin = 5;                  // pin to do fancy classy fading blink at each beat
int fadeRate = 0;                 // used to fade LED on with PWM on fadePin
volatile int IBI = 600;          // holds the time between beats, must be seeded!
volatile int BPM;                // used to hold the pulse rate
volatile int Signal;             // holds the incoming raw data
volatile boolean QS = false;     // becomes true when Arduino finds a beat

void setup()
{
  pinMode(blinkPin,OUTPUT);   // pin that will blink to your heartbeat!
  pinMode(fadePin,OUTPUT);          // pin that will fade to your heartbeat!
  Serial.begin(115200);             // we agree to talk fast!  
  PulseSensor::begin()
}
void loop(){
  sendDataToProcessing('S',Signal);     // send Processing the raw Pulse Sensor data
  if (QS == true){                       // Quantified Self flag is true when arduino finds a heartbeat
        fadeRate = 255;                  // Set 'fadeRate' Variable to 255 to fade LED with pulse
        sendDataToProcessing('B',BPM);   // send heart rate with a 'B' prefix
        sendDataToProcessing('Q',IBI);   // send time between beats with a 'Q' prefix
        QS = false;                      // reset the Quantified Self flag for next time    
     }   
  ledFadeToBeat();
  delay(5000);                             //  take a break
}
void ledFadeToBeat(){
    fadeRate -= 15;                         //  set LED fade value
    fadeRate = constrain(fadeRate,0,255);   //  keep LED fade value from going into negative numbers!
    analogWrite(fadePin,fadeRate);          //  fade LED
  }

void sendDataToProcessing(char symbol, int data )
{
    Serial.print(symbol);                // symbol prefix tells Processing what type of data is coming
    Serial.println(data);                // the data to send culminating in a carriage return
  }

my cpp
//

#include "pulse-sensor-arduino.h"

volatile int rate[10];                        // array to hold last ten IBI values
volatile unsigned long sampleCounter = 0;     // used to determine pulse timing
volatile unsigned long lastBeatTime = 0;      // used to find IBI
volatile int P =512;                          // used to find peak in pulse wave, seeded
volatile int T = 512;                         // used to find trough in pulse wave, seeded
volatile int thresh = 512;                    // used to find instant moment of heart beat, seeded
volatile int amp = 100;                       // used to hold amplitude of pulse waveform, seeded
volatile boolean firstBeat = true;            // used to seed rate array so we startup with reasonable BPM
volatile boolean secondBeat = false;          // used to seed rate array so we startup with reasonable BPM
volatile boolean Pulse = false;               // true when pulse wave is high, false when it's low
volatile int pulsePin;

volatile int PulseSensor::IBI = 600;          // holds the time between beats, must be seeded!
volatile int PulseSensor::BPM;                // used to hold the pulse rate
volatile int PulseSensor::Signal;             // holds the incoming raw data
volatile boolean PulseSensor::QS = false;     // becomes true when Arduino finds a beat

void PulseSensor::begin(int pPin)
{
  
[... 105 lines cut ...]

#include "Arduino.h"

class PulseSensor
{
public:
  static void begin(int pulsePin);

  static volatile int BPM;                               // used to hold the pulse rate
  static volatile int Signal;                        // holds the incoming raw data
  static volatile int IBI;                            // holds the time between beats, must be seeded!
  static volatile boolean QS;                     // becomes true when Arduoino finds a beat.
};

#endif // PULSE_SENSOR_ARDUINO_H

error shown is

///
In function 'void setup()':
 error: '::begin' has not been declared
 note: in expansion of macro 'PulseSensor'
 error: expected ';' before '::' token
Error compiling.
///

ffissore, 2015-02-06:
Why did you `#define PulseSensor::begin()` ?

karthisl26, 2015-02-09:
for calling the library function

ffissore, 2015-02-09:
This is where the error lies, as the error message you posted tells you. I'm therefore closing this as invalid. Please review your sketch. Post your code on the [Arduino forum](http://forum.arduino.cc/index.php?board=4.0) "Programming Questions" section

closed 2015-02-09 by ffissore

karthisl26, 2015-02-09:
thanks for your suggestion