arduino/Arduino #2604: IDE doesn't start after double clicking a lonely sketch

ffissore, 2015-02-06:
As reported on the developers mailing list

Nevada Smith ha scritto il 06/02/2015 alle 15:48:> Found a bug (windows tested):
> Create a sketch and move it out of the normal folder. Double click it. 
> The IDE will ask to create a folder for this sketch. It does, but it 
> doesnt move the sketch into it and the IDE wont start afterwards. Also 
> appeared in 1.6 rc2, dont know any earlier release.

NicoHood, 2015-02-06:
Make sure to test it under other OS as well (had no chance to do). Good Luck ;)

moshi04, 2015-02-11:
Please check the check-sum or integrity error of windows installer file. Its not starting after download showing integrity error. 
[image]

NicoHood, 2015-02-11:
Isnt this a totally different bug?
I am using the zip version anyways.

ffissore, 2015-02-11:
@moshi04 please retry downloading and paste the url you're downloading from

moshi04, 2015-02-11:
@ffissore Thank you very much. Now its work.

NicoHood, 2015-02-24:
Any fix for the actual problem?

ffissore, 2015-02-25:
There is an open bounty on this issue. First one that solves it opening a valid pull request wins 50$

PaulStoffregen, 2015-02-25:
Who's paying the bounty, if it gets fixed?

ffissore, 2015-02-25:
That's about how Bountysource works. Basically, Arduino has already paid the bounty. The winner will see the 50$ coming to her bountysource account, and she'll then decide to use them or cash them out
https://github.com/bountysource/frontend/wiki/Frequently-Asked-Questions#does-100-of-the-bounty-go-to-the-developer

gerito1, 2015-02-26:
This pull request should fix that.
https://github.com/arduino/Arduino/pull/2700

NicoHood, 2015-02-26:
haha if that fixes it, great job.
Did I miss something about the bounty? Do you do this for other issues as well?

PaulStoffregen, 2015-02-26:
Yup, that's the right fix!

ffissore, 2015-02-27:
Indeed it is. Thank you very much!

closed 2015-02-27 by ffissore