arduino/Arduino #2607: Trouble with unoptimised for-loops

robin1106, 2015-02-08:
I am experiencing an issue with a for-loop based search for a string in a table in PROGMEM.
Example:
-----------------------------------------------------------------------------------------------
#include <Arduino.h>
#include <avr/pgmspace.h>

// Useful macro for printing strings from PROGMEM
#define G(s)(reinterpret_cast<const __FlashStringHelper *>(s))

// Whitelist of permitted SMS senders
prog_char wl_HomeADR[]   PROGMEM = "+4588888888";
prog_char wl_AltADR[]    PROGMEM = "+4599999999";
prog_char wl_Alt2ADR[]   PROGMEM = "+4577777777";

PROGMEM const char* whitelist[] =
  {wl_HomeADR, wl_AltADR, wl_Alt2ADR};

boolean inwhitelist(char* buf)
{ // Returns true if buf contains a string which is in the 
  //   whitelist.
  for (uint8_t wlidx=0; wlidx<WLLEN; wlidx++)
    { Serial.print(wlidx); Serial.print(':');
      Serial.println(G(whitelist[wlidx]));
      if (strcmp_P(buf, (prog_char *) whitelist[wlidx]) == 0) return true;
    };
  return false;
}
------------------------------------------------------------------------------------------------

If WLLEN is 2, the for loop is optimised to deal with the two first elements in the whitelist separately.  This works correctly: Serial.println(...) prints the contents of the inspected strings, and the function whitelist returns true or false, depending on whether the buffer buf contains a string which matches one of the whitelist strings.

If WLLEN is 3 (or larger), the for loop code loops round, inspecting one string for each loop iteration.  This does not work correctly: Serial.println(...) prints rubbish (in fact the same rubbish for all values of wlidx), and the function whitelist always returns the value false.

I am using arduino-1.5.4.

matthijskooijman, 2015-02-09:
First off - this tracker isn't for problems in your own code, but for bugs in the IDE and Arduino enviroment. However, since I had to read your issue before realizing it doesn't really belong here, I might as well respond to it :-)

`
PROGMEM const char* whitelist[] =
`

Here, you declare that the `whitelist` array itself should be stored in PROGMEM. This means that if you access any element of the array, you should use `pgm_read_word` (IIRC) to get the real value. Your code uses `whitelist[wlidx]` directly, so that tries to read the value from RAM, using the address that is valid for PROGMEM only.

When `WLLEN` is 2, I guess the for loop is unrolled and the `whitelist` array indexing happens at compiletime, making things work even though the code is really incorrect. I hope this gives you enough pointers to understand what's happening and fix it.

closed 2015-02-09 by matthijskooijman

robin1106, 2015-02-17:
I'm sorry I seem to have expressed myself badly.
I know how it should be done, so it was not help with finding the right C++ code which was the issue.

The problem was that the code generated was inconsistent, depending on how large the PROGMEM array to be searched is.  I have written a number of code generators myself for various target machines, and in my environment it is considered just as wrong to produce (apparently) correct object code from a faulty source text as it is to produce incorrect code from a correct source text.  

The reason for this is that the user (the programmer) in such cases gets no indication that his source text is wrong, unless he/she by chance or experiment discovers the critical size of array or other structure where the optimizer changes strategy -- in the case of my example to go from an unrolled loop to a non-unrolled loop.  So for me this was a fault in the compiler, which I thought you should know about.

I've spent a little time on looking at the generated code in some other cases, and some of them seem to have the same sort of problem.  But I feel from your answer that this is perhaps a problem which lies outside the realm of this tracker -- though I don't know which tracker would be better to approach.

Many regards
  RObin

matthijskooijman, 2015-02-18:
> and in my environment it is considered just as wrong to produce (apparently) correct object code from a faulty source text as it is to produce incorrect code from a correct source text. 

From a C-standard perspective you are wrong: It is allowed for a compiler to produce a working program from code that has undefined behaviour. From a sane-programming-tool perspective, you are of course right: The compiler should warn you about stuff like this.

On thing you could try is to modify platform.txt to remove the "-w" option from the compiler commandline, and add "-Wall". (Alternatively, enable verbose building, and copy-paste the gcc commandline for this source file while making the above modifications). This enables compiler warnings. If that shows an error for your code, then your problem will be fixed as soon as we re-enable warnings again (there's another ticket for that).

If that doesn't show any warnings (which is what I expect), then the compiler is probably not smart enough to note this problem. It's actually tricky for the compiler - the C language doesn't really know about PROGMEM, or distinguish between multiple memory types and consequently doesn't distinguish between pointer types into different memories either. This makes it hard for the compiler to track how a pointer is used (though it's certainly not impossible for it to catch the obviously wrong cases, I'm not sure how hard it would be to implement).

If the compiler indeed does not show an error, you could report this issue at the [gcc bugtracker](https://gcc.gnu.org/bugzilla/). It would be best if you removed the Arduino include, to make the example self-contained, only needing gcc.

robin1106, 2015-02-18:
Fine.  Thanks for your suggestions.  I'll work on them.

Robin