arduino/Arduino #2608: 1.6 rc3 Windows Cant open Serial Monitor after a long time of coding

NicoHood, 2015-02-08:
After some hard time of coding I get this error when I try to open the Serial Monitor on windows:

 Exception in thread "AWT-EventQueue-0" java.lang.OutOfMemoryError: Java heap space

Had this Yesterday too. Before this error another error appeared which disappered then and afterwards this came. (dont have the first error, but it through a lot of java standard ide errors). Reprogramming the Arduino still works finde, jus the Serial monitor strikes.

Switching the port or the board type doesnt help. Restarting the IDE does.

ffissore, 2015-02-08:
Can you share the whole sketch?

NicoHood, 2015-02-08:
The sketch changed over the time. Here is a the last "savestate" i have:
https://github.com/NicoHood/Nintendo/tree/594cb4f14df829ab46f6c8b444aeda7bec544232

ffissore, 2015-02-09:
I'm unable to reproduce the problem. Was autoscroll enable on the serial monitor? Can you add some more details?

NicoHood, 2015-02-09:
Yes it was.

The problem occured after a day of coding. I opened the Serial Monitor many times one both days. And at some point the Serial Monitor did not want to open again. So Autoscroll might not be relevant here. I reprogrammed the Arduino and wanted to open the monitor again. Then I got some longer java errors (which i had no chance to post because i thought its one of the "normal" java bugs) and then this error occured.

I am not sure if this problem was causes by the IDE runtime, the times i Opened the Monitor or some other bugs. All i can say is that I had this two times (and one time I am not sure if it was the same bug, the whole ide crashed) after a long period of coding with the same IDE window (about 12 hours of developing).

ffissore, 2015-02-09:
Argh, that will make it hard to debug. Please remind that there are NO normal bugs. Such a thing like a normal bug does NOT exist. If it exists, it's a bug, and must be fixed.

Closing this issue for now: please reopen it as soon as you can paste an error message or produce a test sketch which makes it easier to reproduce the bug.

closed 2015-02-09 by ffissore

NicoHood, 2015-02-09:
Would it help to start the debug exe to get more information? Then I will code with this one. Or is there any other way to give you more information about those crashes?

Normal bugs: I often get some java errors when coding. I think with 1.0.x it happened more often but people always said that this is "normal" for the IDE.

ffissore, 2015-02-09:
Yes indeed. Open a command prompt, `cd` into the Arduino IDE folder, run `arduino_debug.exe` and start coding. Don't double click on `arduino_debug.exe` because the command prompt will close right after the IDE and you'll lose the error message.

NicoHood, 2015-02-09:
May I also write it directly into a textfile with?

> arduinolog.txt

Maybe you could add this as feature for the debug version, that it logs the stuff into a txt file?

ffissore, 2015-02-09:
That's a nice feature request. Please open a separate issue for it

NicoHood, 2015-02-21:
```
Using library PinChangeInterrupt in folder: C:\Users\Nico\Dropbox\Arduino\sketchbook\libraries\PinChangeInterrupt (legacy)

C:\Users\Nico\Documents\arduino-1.6.0/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10600 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -IC:\Users\Nico\Dropbox\Arduino\sketchbook\hardware\HID\avr\cores\hid -IC:\Users\Nico\Dropbox\Arduino\sketchbook\hardware\HoodLoader2\avr\variants\standard_hid -IC:\Users\Nico\Dropbox\Arduino\sketchbook\libraries\PinChangeInterrupt C:\Users\Nico\AppData\Local\Temp\build4503429386178947383.tmp\PinChangeInterrupt_Api_Basic.cpp -o C:\Users\Nico\AppData\Local\Temp\build4503429386178947383.tmp\PinChangeInterrupt_Api_Basic.cpp.o 
C:\Users\Nico\Documents\arduino-1.6.0/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10600 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -IC:\Users\Nico\Dropbox\Arduino\sketchbook\hardware\HID\avr\cores\hid -IC:\Users\Nico\Dropbox\Arduino\sketchbook\hardware\HoodLoader2\avr\variants\standard_hid -IC:\Users\Nico\Dropbox\Arduino\sketchbook\libraries\PinChangeInterrupt -IC:\Users\Nico\Dropbox\Arduino\sketchbook\libraries\PinChangeInterrupt\utility C:\Users\Nico\Dropbox\Arduino\sketchbook\libraries\PinChangeInterrupt\PinChangeInterrupt.cpp -o C:\Users\Nico\AppData\Local\Temp\build4503429386178947383.tmp\PinChangeInterrupt\PinChangeInterrupt.cpp.o 
Using previously compiled file: C:\Users\Nico\AppData\Local\Temp\build4503429386178947383.tmp\hooks.c.o
Using previously compiled file: C:\Users\Nico\AppData\Local\Temp\build4503429386178947383.tmp\WInterrupts.c.o
Using previously compiled file: C:\Users\Nico\AppData\Local\Temp\build4503429386178947383.tmp\wiring.c.o
Using previously compiled file: C:\Users\Nico\AppData\Local\Temp\build4503429386178947383.tmp\wiring_analog.c.o
Using previously compiled file: C:\Users\Nico\AppData\Local\Temp\build4503429386178947383.tmp\wiring_digital.c.o
Using previously compiled file: C:\Users\Nico\AppData\Local\Temp\build4503429386178947383.tmp\wiring_pulse.c.o
Using previously compiled file: C:\Users\Nico\AppData\Local\Temp\build4503429386178947383.tmp\wiring_shift.c.o
Using previously compiled file: C:\Users\Nico\AppData\Local\Temp\build4503429386178947383.tmp\abi.cpp.o
Using previously compiled file: C:\Users\Nico\AppData\Local\Temp\build4503429386178947383.tmp\HardwareSerial.cpp.o
Using previously compiled file: C:\Users\Nico\AppData\Local\Temp\build4503429386178947383.tmp\HardwareSerial0.cpp.o
Using previously compiled file: C:\Users\Nico\AppData\Local\Temp\build4503429386178947383.tmp\HardwareSerial1.cpp.o
Using previously compiled file: C:\Users\Nico\AppData\Local\Temp\build4503429386178947383.tmp\HardwareSerial2.cpp.o
Using previously compiled file: C:\Users\Nico\AppData\Local\Temp\build4503429386178947383.tmp\HardwareSerial3.cpp.o
Using previously compiled file: C:\Users\Nico\AppData\Local\Temp\build4503429386178947383.tmp\IPAddress.cpp.o
Using previously compiled file: C:\Users\Nico\AppData\Local\Temp\build4503429386178947383.tmp\main.cpp.o
Using previously compiled file: C:\Users\Nico\AppData\Local\Temp\build4503429386178947383.tmp\new.cpp.o
Using previously compiled file: C:\Users\Nico\AppData\Local\Temp\build4503429386178947383.tmp\Print.cpp.o
Using previously compiled file: C:\Users\Nico\AppData\Local\Temp\build4503429386178947383.tmp\Stream.cpp.o
Using previously compiled file: C:\Users\Nico\AppData\Local\Temp\build4503429386178947383.tmp\Tone.cpp.o
Using previously compiled file: C:\Users\Nico\AppData\Local\Temp\build4503429386178947383.tmp\CDC.cpp.o
Using previously compiled file: C:\Users\Nico\AppData\Local\Temp\build4503429386178947383.tmp\Consumer.cpp.o
Using previously compiled file: C:\Users\Nico\AppData\Local\Temp\build4503429386178947383.tmp\Gamepad.cpp.o
Using previously compiled file: C:\Users\Nico\AppData\Local\Temp\build4503429386178947383.tmp\HID.cpp.o
Using previously compiled file: C:\Users\Nico\AppData\Local\Temp\build4503429386178947383.tmp\Keyboard.cpp.o
Using previously compiled file: C:\Users\Nico\AppData\Local\Temp\build4503429386178947383.tmp\Mouse.cpp.o
Using previously compiled file: C:\Users\Nico\AppData\Local\Temp\build4503429386178947383.tmp\RawHID.cpp.o
Using previously compiled file: C:\Users\Nico\AppData\Local\Temp\build4503429386178947383.tmp\System.cpp.o
Using previously compiled file: C:\Users\Nico\AppData\Local\Temp\build4503429386178947383.tmp\USBCore.cpp.o
Using previously compiled file: C:\Users\Nico\AppData\Local\Temp\build4503429386178947383.tmp\WMath.cpp.o
Using previously compiled file: C:\Users\Nico\AppData\Local\Temp\build4503429386178947383.tmp\WString.cpp.o
Using previously compiled file: C:\Users\Nico\AppData\Local\Temp\build4503429386178947383.tmp\core.a
C:\Users\Nico\Documents\arduino-1.6.0/hardware/tools/avr/bin/avr-gcc -w -Os -Wl,--gc-sections -mmcu=atmega328p -o C:\Users\Nico\AppData\Local\Temp\build4503429386178947383.tmp/PinChangeInterrupt_Api_Basic.cpp.elf C:\Users\Nico\AppData\Local\Temp\build4503429386178947383.tmp\PinChangeInterrupt_Api_Basic.cpp.o C:\Users\Nico\AppData\Local\Temp\build4503429386178947383.tmp\PinChangeInterrupt\PinChangeInterrupt.cpp.o C:\Users\Nico\AppData\Local\Temp\build4503429386178947383.tmp/core.a -LC:\Users\Nico\AppData\Local\Temp\build4503429386178947383.tmp -lm 
[... cut ...]
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue.dispatchEvent(Unknown Source)
Exception in thread "AWT-EventQueue-0" java.lang.OutOfMemoryError: Java heap space
Exception in thread "AWT-EventQueue-0" java.lang.OutOfMemoryError: Java heap space

```

I should note that I printed quite a lot while developing. Also its not even that much I'd say.
Used Arduino Uno win8. IDE 1.6 again.

I dont get any more information, thats all I have. **Please reopen the issue.**

Shottgun, 2022-10-21:
For anyone else getting this 7 years after it was posted, restarting my computer solved it :D