arduino/Arduino #2610: Tinkerkit DMX Receiver - No documentation - Arduino Library

maxxist, 2015-02-08:
I purchased a Tinkerkit DMX Receiver - Relay and I can not find documentation or
the DMX library needed to run on an Arduino uno board.
Thanks
maxxist

sxpert, 2015-02-13:
the tinkerkit.com website appears dead...

q2dg, 2015-04-04:
Maybe this could help? https://github.com/TinkerKit/DmxMaster

maxxist, 2015-04-04:
Thanks
MP

q2dg, 2015-04-04:
Well, I would like to insist in this issue.

There's no bit of information about what happened to Tinkerkit. In the forum reigns the silence (http://forum.arduino.cc/index.php?topic=261238.0). In Arduino Store there's no advice neither. Last updates in its Github come from 2013 (except an strange commit from two month ago)...

If official boards like Esplora and official shiedls like Motor and USBHost carry these kind of connectors, as a affected part Arduino should move on. This is not acceptable situation at all.

mbanzi, 2016-02-29:
HI the TinkerKit Products are NOT Arduino.cc products. We used to sell them and used to add tinkerkit connectors to some products because they were generic enough 3pin connectors. You should contact the manufcturer indicated on the box

closed 2016-02-29 by mbanzi