arduino/Arduino #2611: Official EthernetClient Sample Not Compiling - Ambiguous overload error with latest Arduino 1.0.6

OzmoOzmo, 2015-02-09:
The code sample below - the official Arduino sample code.
will not compile:

        http://arduino.cc/en/Reference/ServerAvailable

My public code libraries use the EthernetClient library like this - and same method I used is mentioned in several books and several ethernet sources - but it will not compile anymore on 1.0.6

(I have tried official 1.0.6 and the nightly build ARDUINO 1.6.0rc3 - 2015.02.03)

The problem is EthernetClient cannot be tested for true/false anymore (it used to have a overloading that allowed you to query it being ready by using boolean comparitor). 

The sample code has not worked since 1.0.5

Official Sample Code: 

  EthernetClient client = server.available();
  if (client == true) {  //compiler error here
  }

What is the correct new way to do this test now - and will the documentation be changed or the IDE/Compiler be changed back (best option)?

Thanks very much.

ffissore, 2015-02-10:
Hi, the code is wrong. We'll fix it asap. Correct syntax is `if (client) {`

OzmoOzmo, 2015-02-10:
I think might be better fix in compiler if possible than update documentation...

Several printed media is currently showing the way it worked in 1.0.5

eg. page 163 of this book.

https://books.google.co.uk/books?id=yvclBgAAQBAJ&pg=PA163&lpg=PA163&dq=arduino+"if+%28client+%3D%3D+true%29"

And this book (Non-English language)
https://books.google.co.uk/books?id=LwLzAwAAQBAJ&pg=PA185&dq=arduino+"if+%28client+%3D%3D+true%29"

ffissore, 2015-02-10:
You're right. I've proposed a change in #2618. Please leave your comment

ffissore, 2015-02-11:
Please try one of the IDEs at https://github.com/arduino/Arduino/pull/2618#issuecomment-73862925

agdl, 2015-02-20:
@OzmoOzmo corrected in the reference. Can i close this issue?

OzmoOzmo, 2015-02-20:
Thanks very much - that does indeed sort that problem.

Sorry not getting back to you sooner - but I was trying to get the whole
project compiling to fully test it
- still having trouble converting depreciated memory functions like
const prog_uchar allmonths[] = {"JANFEB..."} etc.

But thanks for getting EtherClient working anyway :)

PR #2618 (2015-02-23): EthernetClient == and != operators
Added `virtual bool operator==(const bool value)` and `virtual bool operator!=(const bool value)`.
Fixes #2611
Bug introduced with #1700
@cmaglie @matthijskooijman you're far more expert than me with c++
commit eb0c2c9b (2015-02-23): Added virtual bool operator==(const bool value) and virtual bool operator!=(const bool value).
Fixes #2611
Bug introduced with #1700
closed 2015-02-23 by ffissore