arduino/Arduino #2613: Meta Issue: need for more effective management of issues

q2dg, 2015-02-09:
Hello. 
It's obvious that more than 100 PR and more than 800 open issues are a confirmation about something goes wrong in Arduino development's management. Maybe it's the lack of people involved, maybe it's the lack of a protocol...I don't know, but I wanted to share this thought with the community to find among all of us a more constructive way of managing all these issues, merging pull requests, etc. 
Thanks.

ffissore, 2015-02-09:
Best thing you can do is to help us but squashing some bug reported in some issue. Thanks in advance :)

q2dg, 2015-02-10:
Yes, I know. Unfortunately, my Arduino's skill are very limited...i'm only an casual user with bare programming/hardware knowledge. I didn't explain myself: my request was focused on the need of some internal reorganization inside Arduino company. Besides voluntary work, I think Arduino core team would have to think about how to manage the amount of work so many boards carry, and have a plan. Maybe hiring someone (I guess there's money for that), or participating in GSoC or similar iniciatives, or delegating task to different comissions...I don't know. The subject is guarantee the future scalability of the Arduino project as a whole

ffissore, 2015-02-10:
Thanks for the heads up. We really are already doing all that stuff. And we will do that more and more in the coming months. Stay tuned
Closing as invalid, since the issue is not related to the IDE

closed 2015-02-10 by ffissore

PaulStoffregen, 2015-02-11:
There is a school of thought that say you can never actually reduce the number of known bugs in a large piece of software.  As you fix problems, more are discovered.

But as you make progress, the severity and impact of the problems gradually decreases....