arduino/Arduino #2620: IDE does not accept international characters (OS X, macosx-java-latest)

igrr, 2015-02-10:
I'm on OS X 10.10 and I have built the 1.6.0rc3 version in two configurations:
1) with platform=macosx and Apple Java 6
2) with platform=macosx-java-latest and Oracle Java 7
When the IDE is build in the first configuration I can enter non-latin characters in the sketch.
In the second configuration, there is no output (no characters appear, cursor doesn't move) when I enter non-latin characters. I tried with Chinese (pinyin) and Russian keyboard layouts.

ffissore, 2015-02-11:
Using a 10.8 I can type russian chars with both `platform`s. I'll look for a 10.10

ffissore, 2015-02-11:
I confirm the issue using 10.10. Thanks for the heads up, we'll try to fix asap

ffissore, 2015-04-23:
Issue is being fixed with a brand new editor. See #3019

ffissore, 2015-05-07:
#3019 merged. Next hourly build and IDE 1.6.5 will contain the fix

closed 2015-05-07 by ffissore