arduino/Arduino #2621: Migrate wiki pages from google code to github

ffissore, 2015-02-11:
Starting from https://code.google.com/p/arduino/wiki/BuildingArduino because we lack build instructions and they were asked [here](https://github.com/arduino/Arduino/pull/2590#issuecomment-73686622) and [here](https://github.com/arduino/Arduino/issues/2596#issuecomment-73360386)

ffissore, 2015-02-11:
And [here](https://github.com/arduino/Arduino/issues/2160)

damellis, 2015-02-11:
BTW, you might also consider making a landing page of some sort for Arduino development in general, e.g. as the main wiki page: https://github.com/arduino/Arduino/wiki. Just a place for pointers to the build instructions and other wiki pages, developers mailing list, etc.

agdl, 2015-04-03:
@damellis now the building process is here

https://github.com/arduino/Arduino/wiki/Building-Arduino

agdl, 2015-06-12:
Migrated here:
https://github.com/arduino/Arduino/wiki/Build-Process
https://github.com/arduino/Arduino/wiki/Building-avrdude-for-Arduino
https://github.com/arduino/Arduino/wiki/Development-Policy
https://github.com/arduino/Arduino/wiki/Git-Guide-for-Arduino

ffissore, 2015-06-25:
Thanks @agdl. FYI I've deleted page "building avrdude" since there is executable documentation at https://github.com/arduino/toolchain-avr.

closed 2015-06-25 by ffissore