arduino/Arduino #2622: 1.6 Include Arduino.h differs if global int was used

NicoHood, 2015-02-11:
Its a very weird bug. Feel free to edit the name of the issue, I had no better idea.

The Arduino include definition works after a global int was placed in the sketch. it does not work if it is above. If i force the include it works. Simply try the following code and comment the integer:

```cpp
//#include "Arduino.h"

#ifdef Arduino_h
#error
#endif

int nowincludingbewlo = 2;

#ifdef Arduino_h
#error
#endif

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

Can anyone explain me why this happens, so I can at least continuing developing without being afraid that the next bugfix will break my stuff? This confused me a lot and also caused my sketch to now work anymore for some reason. Using new 1.6 with Windows.

ffissore, 2015-02-11:
I think this is a dup of #1518

NicoHood, 2015-02-11:
Wow. that bug is old. Why dont you (or anyone else) fix it? That really kills a lot of stuff and confuses a lot. I spent an hour to find out that my problem wasnt caused by me.

Any chance to get a fix now?

ffissore, 2015-02-11:
I confirm this is a preprocessor related issue and it's being tracked in several preprocessor related bugs: https://github.com/arduino/Arduino/issues?q=is%3Aopen+is%3Aissue+label%3A%22Component%3A+Preprocessor%22

Please try one of these builds
http://arduino.cc/download.php?f=/arduino-preproc-linux32.tar.xz
http://arduino.cc/download.php?f=/arduino-preproc-linux64.tar.xz
http://arduino.cc/download.php?f=/arduino-preproc-windows.zip
http://arduino.cc/download.php?f=/arduino-preproc-macosx.zip
http://arduino.cc/download.php?f=/arduino-preproc-macosx-java-latest.zip

ffissore, 2015-02-13:
New preprocessor tracked at #2636. Builds for testing it are available

NicoHood, 2015-02-15:
```

Build options changed, rebuilding all

processing.app.debug.RunnerException
	at processing.app.debug.Compiler.compile(Compiler.java:352)
	at processing.app.debug.Compiler.build(Compiler.java:98)
	at processing.app.Sketch.build(Sketch.java:1159)
	at processing.app.Sketch.build(Sketch.java:1134)
	at processing.app.Editor$BuildHandler.run(Editor.java:1936)
	at java.lang.Thread.run(Unknown Source)
Caused by: processing.app.helpers.PreferencesMapException: pattern
	at processing.app.helpers.PreferencesMap.getOrExcept(PreferencesMap.java:255)
	at processing.app.preproc.IncludesFinder.preprocess(IncludesFinder.java:43)
	at processing.app.debug.Compiler.compile(Compiler.java:350)
	... 5 more

```

When I try to compile it with my own hid core. Just pasted in my sketch above.

ffissore, 2015-09-11:
Fixed by #3779

closed 2015-09-11 by ffissore