arduino/Arduino #2625: defines not working correct in .ino file

NicoHood, 2015-02-11:
```cpp
void setup() {

}

void loop() {

}

#ifndef thisisnevereverdefined

#else
//this error is displayed in line 4??
wordthatshouldnotcompile
#endif

void thisEmptyMethodeCausesTheError(void)
{

}
```

     sketch_feb11a.ino:4:1: error: 'wordthatshouldnotcompile' does not name a type Error compiling.

using 1.6, Windows. Could be similar to this:
https://github.com/arduino/Arduino/issues/2622#issuecomment-73855653

ffissore, 2015-02-11:
This is being tracked in several preprocessor related bugs: https://github.com/arduino/Arduino/issues?q=is%3Aopen+is%3Aissue+label%3A%22Component%3A+Preprocessor%22

Please try one of these builds
http://arduino.cc/download.php?f=/arduino-preproc-linux32.tar.xz
http://arduino.cc/download.php?f=/arduino-preproc-linux64.tar.xz
http://arduino.cc/download.php?f=/arduino-preproc-windows.zip
http://arduino.cc/download.php?f=/arduino-preproc-macosx.zip
http://arduino.cc/download.php?f=/arduino-preproc-macosx-java-latest.zip

ffissore, 2015-02-13:
New preprocessor tracked at #2636. Builds for testing it are available

ffissore, 2015-09-11:
Fixed by #3779

closed 2015-09-11 by ffissore