arduino/Arduino #2630: conditional processing interacts oddly with includes and variable definitions

plocher, 2015-02-12:
TL;DR simplification of a more complicated real example - It appears as if the IDE compile-time code that processes the sketch to handle automatic function declarations injects the results immediately after the last \#include, even if there are conditional compilation directives in use:

```C
// #define HASLCD
#include <Wire.h>

#ifdef HASLCD
#include <LiquidTWI.h>
#endif

// int x;         // to work around the problem, intersperse some non-#include/non-#define solids to chew on,
// int foo(void); // or avoid the need for any automagic pre-declarition stuff by explicitly predeclaring the function(s)...

#ifdef HASLCD
// The compile problem happens here, because the automagically created pre-declarations are injected 
// inside this #ifdef - which in this scenario will be eliminated...
LiquidTWI lcd(0);  // Set the LCD I2C address
#endif
void setup() {
#ifdef HASLCD
    lcd.begin(20,4);               // initialize the lcd 
    lcd.setBacklight(HIGH);
    lcd.home();
    lcd.print(ME);
#endif

    Wire.begin();
}

void loop() {
    int x = foo();
}
// defined after use with no explicit declaration, relies on IDE automagic for pre-declaration...
int foo(void) {
      return 1;
}
```
becomes (in /var/folders/nw/..../sketch.cpp)
```C
#line 1 "include_define_error.ino"
#include <Wire.h>
#ifdef HASLCD
#include <LiquidTWI.h>
#endif
#ifdef HASLCD
#include "Arduino.h"
void setup();
void loop();
int foo(void);
#line 14
LiquidTWI lcd(0);                            
#endif
```
and results in (as expected...) compile failures because any of the functions defined after loop() have no associated declarations in scope:
```C
include_define_error.ino: In function 'void loop()':
include_define_error.ino:36:17: error: 'foo' was not declared in this scope
Error compiling.
```
Obviously, pre-declaring the functions explicitly, or ordering the definitions and uses in such a way as to avoid the need for explicit forward declarations is the "proper" way to solve the problem - but that begs the question of the value of automagically computing declarations...  In the spirit of making the automagic work, I moved the conditional variable declaration AFTER some other variable declaration/definition, which seems to patch things back up...

Yeah, I know the IDE handles conditionals poorly, but this type of silent failure is effectively impossible for the most users to debug...

ffissore, 2015-02-12:
Yep, current preprocessor is [quite weak](https://github.com/arduino/Arduino/issues?q=is%3Aopen+is%3Aissue+label%3A%22Component%3A+Preprocessor%22), but we already have an alternative

Give a spin to this builds of the IDE
http://arduino.cc/download.php?f=/arduino-preproc-linux32.tar.xz
http://arduino.cc/download.php?f=/arduino-preproc-linux64.tar.xz
http://arduino.cc/download.php?f=/arduino-preproc-windows.zip
http://arduino.cc/download.php?f=/arduino-preproc-macosx.zip
http://arduino.cc/download.php?f=/arduino-preproc-macosx-java-latest.zip

Code hasn't been merged yet because is hasn't received enough feedback: it's a change to a fundamental piece, needs as much feedback as possible

ffissore, 2015-02-12:
Your sketch, preprocessed, becomes https://gist.github.com/ffissore/6fbc15f371e8265a6c5e

ffissore, 2015-02-12:
With the #define uncommented, it becomes https://gist.github.com/ffissore/59aead0a302c23dfdd9d

ffissore, 2015-02-13:
New preprocessor tracked at #2636. Builds for testing it are available

plocher, 2015-02-17:
Verified fix on MacOS, with the supplied test build.  Thank you!

  -John

On Fri, Feb 13, 2015 at 10:45 AM, Federico Fissore <notifications@github.com
> wrote:

> New preprocessor tracked at #2636
> <https://github.com/arduino/Arduino/pull/2636>. Builds for testing it are
> available
>

ffissore, 2015-09-11:
Fixed by #3779

closed 2015-09-11 by ffissore