arduino/Arduino #2631: DUE Serial flush()

TeamSouth, 2015-02-12:
I'm currently using a DUE with IDE version 1.6.0, and I'm having trouble with the Serial1.flush()
 returning before the Serial1 Tx buffer is empty. My project code is very extensive, but here is a snippet of what I'm trying to do. Also, the problem is intermittent and never occurs in the same place during the Serial1.print() or Serial1.println(). Any help would be greatly appreciated. On a side not, there are a fair amount of Serial1.flush() statements in the code due to the fact that I was trying to troubleshoot where the error was occurring.

bool Magnetometer::getWaypoints()
{
  bool done = false;

  while ( Serial1.read() != -1 );//clear incoming buffer

  Serial1.println("");
  Serial1.print("Please Enter The Number Of Waypoints You Desire(NOT TO EXCEED 20): ");
  Serial1.flush();
  delay(10);

  while (!Serial1.available());//wait until user inputs data
  delay(100);
  input = received();//read in user input

  Serial1.println(input);

  maxWP = input.toInt();//converts string input to an integer

  //initializes waypoints
  for (j = 0; j < 2; j++)
    for (i = 0; i < 20; i++)
      waypoints[j][i] = 0;

  Serial1.println("The Format For Entering Waypoints is: LAT:DDDMM.SSSS,LONG:DDDMM.SSSS");
  Serial1.flush();
  Serial1.println("For Example, Enter: DDDMM.SSSS,DDDMM.SSSSS");
  Serial1.flush();
  delay(10);

  //this block reads in the number of desired waypoints
  for (i = 0; i < maxWP; i++)
  {
    memset(buf, 0, sizeof(buf));
    memset(buf2, 0, sizeof(buf));

    while ( Serial1.read() != -1 );//clear incoming buffer

    Serial1.println("");
    Serial1.print("Please Enter Wayopint ");
    Serial1.flush();
    Serial1.print((i + 1));
    Serial1.flush();
    Serial1.print(" : ");
    Serial1.flush();

    while (!Serial1.available());//wait until user inputs data
    delay(100);
    input = received();//read in user input
    Serial1.print(input);
    Serial1.flush();
    delay(10);

    //parse user input into waypoint array
    input.toCharArray(buf, sizeof(buf));//converts string to a char array
    getWpField(buf2, 0); //gets LAT
    waypoints[0][i] = atof(buf2);//converts string input to an integer
    getWpField(buf2, 1); //gets LONG
    waypoints[1][i] = atof(buf2);//converts string input to an integer

  }//end of for

  //prints out waypoint array
  Serial1.println();
  Serial1.println("Varifying Waypoints...");
  Serial1.flush();
  delay(100);

  for (i = 0; i < maxWP; i++)
  {
    Serial1.print("Waypoint ");
    Serial1.flush();
    Serial1.print((i + 1));
    Serial1.flush();
    Serial1.print(": ");
    Serial1.flush();
    Serial1.print(waypoints[0][i], 4);
    Serial1.flush();
    Serial1.print(",");
    Serial1.flush();
    Serial1.println(waypoints[1][i], 4);
    Serial1.flush();
    delay((100 * maxWP));

  }//end of outter for

  bool WpTest = false;

  do
  {
    while ( Serial1.read() != -1 );//clear incoming buffer

    Serial1.println();
    Serial1.flush();
    Serial1.print("Are The Waypoints Correct? (Y or N): ");
    Serial1.flush();
    delay(100);

    while (!Serial1.available());//wait until user inputs data
    delay(100);

    input = received();//read in user input
    input.toCharArray(buf, 2);//converts string to a char array
    input = String(buf);
    input.toUpperCase();
    Serial1.println(input);
    Serial1.flush();
    delay(10);

    if (input == "Y" || input == "N")
      check = valid;
    else
      check = invalid;

    switch (check)
    {
      case valid:
        WpTest = true;
        break;
      case invalid:
        Serial1.println("Invalid Entry! Please Enter Y or N...");
        Serial1.flush();
        delay(10);
        WpTest = false;
        delay(100);
        break;
    }

  } while (!WpTest);

  if (input == "Y")
    done = true;
  if (input == "N")
    done = false;

  return done;

}//end of getWaypoints

closed 2015-02-16 by TeamSouth

cmaglie, 2015-02-16:
@TeamSouth 

have you closed this one because you solved the issue?

TeamSouth, 2015-02-17:
Basically. The issue is that the data can't be pushed at 115.2 kbps due to the Arduino IDE overhead. 

Sent from Yahoo Mail on Android