arduino/Arduino #2632: 1.6 Windows: Java error when Serial opened by another instance

NicoHood, 2015-02-12:
Open an IDE, start Serial monitor.
Open another IDE instance, try to upload to this port. Java errors will appear:

```
Sketch uses 7,140 bytes (24%) of program storage space. Maximum is 28,672 bytes.
Global variables use 363 bytes (35%) of dynamic memory, leaving 661 bytes for local variables. Maximum is 1,024 bytes.
Forcing reset using 1200bps open/close on port COM16
processing.app.debug.RunnerException
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(Unknown Source)
	at processing.app.debug.Compiler.upload(Unknown Source)
	at processing.app.Sketch.upload(Sketch.java:1218)
	at processing.app.Sketch.exportApplet(Sketch.java:1191)
	at processing.app.Sketch.exportApplet(Sketch.java:1163)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2391)
	at java.lang.Thread.run(Unknown Source)
Caused by: processing.app.SerialException: Error touching serial port 'COM16'.
	at processing.app.Serial.touchPort(Unknown Source)
	... 7 more
Caused by: jssc.SerialPortException: Port name - COM16; Method name - openPort(); Exception type - Port busy.
	at jssc.SerialPort.openPort(SerialPort.java:164)
	... 8 more
```

ffissore, 2015-02-12:
The error is expected. It does look scary though

NicoHood, 2015-02-12:
What do you mean by expected? That a normal user will run into this problem very often? At least I did now, that I was so sick of it, posting it right now :D

ffissore, 2015-02-12:
It's expected because you can't open the same serial device more than once. As said, the error message could be less scary

PR #2633 (2015-02-24): Better error message when opening serial monitor on a busy serial device
Closes #2632

@cmaglie @NicoHood
commit e385f67f (2015-02-24): Better error message when opening serial monitor on a busy serial device. Closes #2632
closed 2015-02-24 by ffissore