arduino/Arduino #2634: functions no longer return what subfunctiosn return to them

SEKCobra, 2015-02-12:
Before yesterday I was using 1.0.6, and many of the libraries I am using have functions which call another function and have a returntype, but no explicit return. They returned the value of the function inside them automatically, but with 1.6 it no longer does that, it just always returns 0. Is there any way to at least get a warning about this? Like returntype defined but null returned.

ffissore, 2015-02-12:
Can you add a link to one of these libraries or better share a sample sketch?

matthijskooijman, 2015-02-12:
I suspect that the previous behaviour was just accidentally caused by the way the compiler worked. Not returning anything in a non-void functions is, AFAIK, undefined behaviour (or perhaps unspecfied behaviour0 according to the C specification. This means that the result is not defined - with previous compiler versions, it just accidentally worked, probably because the return value from the other function was still lying around in the return value register and didn't get overwritten. Perhaps the new compiler shipped with 1.6 explicitely clears out the return value, or uses the register for something else after the function call.

In any case, if there is code where a non-void function does not contain an explicit `return` statement, that code is broken and should be fixed.

Ideally, you'd get a warning from the compiler about this, but those are still disabled (see #1728).

SEKCobra, 2015-02-12:
http://upstore.net/UEsBzw

matthijskooijman, 2015-02-12:
Heh, lot of hassle to download a zip with a single file from that site. Here's the sketch contained in it:

```
#define TIMERCOUNT 1
unsigned long timer[TIMERCOUNT];
byte timerState[TIMERCOUNT];

void setup() {
  Serial.begin(9600);
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:
  Serial.println(delaySeconds(0,1));
  
  delay(50);

}

int delayHours(byte timerNumber, unsigned long delaytimeH) {  //Here we make it easy to set a delay in Hours
  return delayMilliSeconds(timerNumber, delaytimeH * 1000 * 60 * 60);
}
int delayMinutes(byte timerNumber, unsigned long delaytimeM) {  //Here we make it easy to set a delay in Minutes
  return delayMilliSeconds(timerNumber, delaytimeM * 1000 * 60);
}
int delaySeconds(byte timerNumber, unsigned long delaytimeS) {  //Here we make it easy to set a delay in Seconds
  return delayMilliSeconds(timerNumber, delaytimeS * 1000);
}
int delayMilliSeconds(int timerNumber, unsigned long delaytime) {
  unsigned long timeTaken;
  if (timerState[timerNumber] == 0) { //If the timer has been reset (which means timer (state ==0) then save millis() to the same number timer,
    timer[timerNumber] = millis();
    timerState[timerNumber] = 1;    //now we want mark this timer "not reset" so that next time through it doesn't get changed.
  }
  if (millis() > timer[timerNumber]) {
    timeTaken = millis() + 1 - timer[timerNumber]; //here we see how much time has passed
  }
  else {
    timeTaken = millis() + 2 + (4294967295 - timer[timerNumber]); //if the timer rolled over (more than 48 days passed)then this line accounts for that
  }
  if (timeTaken >= delaytime) {        //here we make it easy to wrap the code we want to time in an "IF" statement, if not then it isn't and so doesn't get run.
    timerState[timerNumber] = 0; //once enough time has passed the timer is marked reset.
    return 1;                          //if enough time has passed the "IF" statement is true
  }
  else {                               //if enough time has not passed then the "if" statement will not be true.
    return 0;
  }
}
```

@SEKCobra, I don't see any functions without an explicit return statement. Can you clarify?

SEKCobra, 2015-02-12:
Sorry about that, couldn't find any filehosts that don't require logins anymore.

Also, I just realized that I fixed that one up while testing it, the return of all functions except delayMilliSeconds() was added by me.

SEKCobra, 2015-02-12:
So Like this:
```
#define TIMERCOUNT 1
unsigned long timer[TIMERCOUNT];
byte timerState[TIMERCOUNT];

void setup() {
  Serial.begin(9600);
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:
  Serial.println(delaySeconds(0,1));

  delay(50);

}

int delayHours(byte timerNumber, unsigned long delaytimeH) {  //Here we make it easy to set a delay in Hours
  delayMilliSeconds(timerNumber, delaytimeH * 1000 * 60 * 60);
}
int delayMinutes(byte timerNumber, unsigned long delaytimeM) {  //Here we make it easy to set a delay in Minutes
  delayMilliSeconds(timerNumber, delaytimeM * 1000 * 60);
}
int delaySeconds(byte timerNumber, unsigned long delaytimeS) {  //Here we make it easy to set a delay in Seconds
  delayMilliSeconds(timerNumber, delaytimeS * 1000);
}
int delayMilliSeconds(int timerNumber, unsigned long delaytime) {
  unsigned long timeTaken;
  if (timerState[timerNumber] == 0) { //If the timer has been reset (which means timer (state ==0) then save millis() to the same number timer,
    timer[timerNumber] = millis();
    timerState[timerNumber] = 1;    //now we want mark this timer "not reset" so that next time through it doesn't get changed.
  }
  if (millis() > timer[timerNumber]) {
    timeTaken = millis() + 1 - timer[timerNumber]; //here we see how much time has passed
  }
  else {
    timeTaken = millis() + 2 + (4294967295 - timer[timerNumber]); //if the timer rolled over (more than 48 days passed)then this line accounts for that
  }
  if (timeTaken >= delaytime) {        //here we make it easy to wrap the code we want to time in an "IF" statement, if not then it isn't and so doesn't get run.
    timerState[timerNumber] = 0; //once enough time has passed the timer is marked reset.
    return 1;                          //if enough time has passed the "IF" statement is true
  }
  else {                               //if enough time has not passed then the "if" statement will not be true.
    return 0;
  }
}
```

matthijskooijman, 2015-02-12:
Right. That code is broken, the return values of those functions are undefined (i.e. anything is allowed to the compiler). If it worked, that's just luck. You need to fix the code.

The underlying problem is that the compiler didn't tell you about this, but we already have #1728 for that. 

I think we can close this issue?

SEKCobra, 2015-02-12:
I agree, that is one glaring issue tho.

closed 2015-02-12 by ffissore

ffissore, 2015-04-10:
I've added a new preference "show all compiler warnings" that turns warnings on
Available as nightly builds in the next few hours