arduino/Arduino #2635: NPE & unable to start if selected board doesn't exist

PaulStoffregen, 2015-02-12:
If you add a 3rd party package, select one of its boards, and then quit Arduino and delete the package, the IDE will be unable to start again.

I believe the problem is likeky in BaseNoGui.onBoardOrPortChange, at line 692:
https://github.com/arduino/Arduino/blob/master/arduino-core/src/processing/app/BaseNoGui.java#L692

      String core = getBoardPreferences().get("build.core");
      if (core.contains(":")) {

If the currently selected board, according to stored preferences, no longer exists, then the string will be null.  Calling contains(":") without first checking for null throws a NPE, and the IDE can't start up.

ffissore, 2015-02-12:
Confirm. If user has a boards.txt that misses `build.core`, an NPE is thrown

closed 2015-02-12 by ffissore

ffissore, 2015-02-13:
Fixed in nightly builds

commit e76d4134 (2015-02-12): When build.core is not defined, fall back to "arduino". Fixes #2635