arduino/Arduino #2638: Missing cmd-line support for Arduino IDE for Java 7 on Mac OS X

der-onkel, 2015-02-12:
Hi there,

i've tried to use the CMD-line tool as described on https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-from-command-line
But it doesn't work on my installation. :-/

kind regards 
Juergen

ffissore, 2015-02-13:
Confirm. cmd line doesn't work on mac :(
Schedule for 1.6.1 is already tight: it will be fixed with 1.6.2

PaulStoffregen, 2015-02-13:
What is the schedule for 1.6.1 ?!

carlosperate, 2015-02-13:
Does the cmd line interface work in 1.5.8 in mac OS X?  What would be the expected schedule for 1.6.2?

carlosperate, 2015-02-22:
Does the command line support at least work on the 1.6 release for Java 6?

ffissore, 2015-02-24:
Yes

ffissore, 2015-03-04:
Fixed with aca6c223ded5c6f718fd4ea14b9b29bc03935549. Available since next nightly and in 1.6.1

closed 2015-03-04 by ffissore

carlosperate, 2015-03-04:
Great to hear, thanks Federico!