arduino/Arduino #2640: core.jar missing from Arduino 1.6.0.

orgicus, 2015-02-13:
The latest stable release(1.6.0) is missing the Processing library (core.jar) which breaks tools using PApplet and other classes within it such as [Gamer Arduino Tools(https://github.com/twsu/GamerArduinoTools).

Is there a chance for this library to ship back with the Arduino 1.6.0 (and future versions) ?

Thank you,
George

ffissore, 2015-02-16:
In order to provide a better command line, IDE has been refactored into `app` and `arduino-core` which is the old `core`
In order to avoid conflicts between classes, `PApplet` was moved from `processing.core.PApplet` to `processing.app.legacy.PApplet`

orgicus, 2015-02-16:
Hi Federico,

Thanks for the heads-up on this. I also notice the legacy version of PApplet is very stripped and there are no renderers(PGraphicsJava2D, etc.). Any recommendations on how to work around this limitation ?

ffissore, 2015-02-16:
@orgicus Are you using the linux version? If so, I may ask you to grab core.jar from IDE 1.5.8 and simply put into `lib` folder or 1.6.0, along with the other jars

orgicus, 2015-02-16:
At the moment I'm on OSX but will need to test on other operating systems later too.
Simply copying core.jar from 1.5.8 to 1.6.0 results in this:
```
Uncaught exception in main method: java.lang.NoClassDefFoundError: processing/core/PApplet
```
Copying core.jar to the tool's folder (alongside the tool .jar) seems to fix the above issue (which stops Arduino from working), but also seems to create some issues with the class paths, because the graphics related classes aren't found (although PApplet is):
```
Exception in thread "AWT-EventQueue-0" java.lang.RuntimeException: You need to use "Import Library" to add processing.core.PGraphicsJava2D to your sketch.
	at processing.core.PApplet.makeGraphics(Unknown Source)
	at processing.core.PApplet.init(Unknown Source)
	at com.techwillsaveus.gamer.Painter.setupSketch(Painter.java:50)
	at com.techwillsaveus.gamer.Painter.run(Painter.java:75)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:209)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:715)
	at java.awt.EventQueue.access$400(EventQueue.java:82)
	at java.awt.EventQueue$2.run(EventQueue.java:676)
	at java.awt.EventQueue$2.run(EventQueue.java:674)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.AccessControlContext$1.doIntersectionPrivilege(AccessControlContext.java:86)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:685)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:296)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:211)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:196)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:188)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:122)
```

I thought I could get away with a quick library copy.
Do you have other simple suggestions before I dig deeper into this ?

ffissore, 2015-02-16:
Nope I'm sorry

orgicus, 2015-02-16:
No worries man, just making sure I tick the simple things first :)

ffissore, 2015-07-02:
Closing as wontfix. New jar name is `arduino-core.jar`. Tools should be recompiled against this new jar and its new package names

closed 2015-07-02 by ffissore

orgicus, 2015-07-02:
The issue is the new jar and new package names don't include the PGraphics classes I need to use. Additionally there are some weird java issues having two version of PApplet (arduino-core.jar and the original PApplet with PGraphics support). 

This is not an issue of me linking against the wrong library,
this is an issue of arduino-core.jar not only removing PGraphics classes, but not renaming
class so a core.jar Processing library can be used alongside arduino-core.jar without conflicts

ffissore, 2015-07-02:
I'm sorry to say we moved away from processing code base a lot during these 3 years and we don't plan to move back and turn compatible again. In essence: a processing tool should be used with processing

orgicus, 2015-07-02:
I understand why, I'm just wondering about how.
Is it possible to refactor/rename the PApplet classes used by arduino-core to different names so they will no longer collide with core.jar if both libraries are used in a tool ?

Regarding your suggestion about the Processing tool, that is what the Tool originally was, 
and then it felt nicer to connect to the Arduino IDE (and the board/port settings, using the editor, and triggering a sketch upload). 
I have a suggestion that might work with both setups: allowing a Tool to be triggered through a socket (TCP or maybe even OSC ). This way, Tools can be developed in Processing or any other language, but can connect to the Arduino IDE and trigger Editor changes, sketch uploads, etc.