arduino/Arduino #2641: 1.6 error in comment handling inside of strings

dieterm57, 2015-02-13:
Following code does not compile:
code:
```
  Serial.println("/\\");
  Serial.println("//");
```

but insert a comment in between:
```
  Serial.println("/\\");
  //\
  Serial.println("//");
```
... and it does
wtf?

matthijskooijman, 2015-02-14:
Could you provide the compiler error messages you are getting?

dieterm57, 2015-02-14:
[image]

```
sketch_feb13b.ino:10:3: error: missing terminating " character
sketch_feb13b.ino: In function 'void loop()':
sketch_feb13b.ino:11:1: error: expected primary-expression before '}' token
Error compiling.
```

ffissore, 2015-02-14:
This is yet another bug related to the preprocessor. Please try one of the builds from #2636

ffissore, 2015-09-11:
Fixed by #3779

closed 2015-09-11 by ffissore