arduino/Arduino #2643: Document other build path properties

PaulStoffregen, 2015-02-14:
The "Build Process" section of the hardware spec documents 2 global properties.

https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5---3rd-party-Hardware-specification#build-process

It should probably also document these:

    build.arch
    build.core.path
    build.system.path
    build.variant.path

VaclavSal, 2015-02-14:
OT 
Paul, thanks for this post, if I only knew that BOSSAC is a 3rd party "hardware" we would still be friends.
Vaclav .

agdl, 2015-02-27:
@PaulStoffregen how sould you document this? At the moment i modified the page indicating what buil.arch identify and specifing that the other you inserted are automatically computed by the IDE. Please take a look and tell me what do you want to improve

agdl, 2015-03-27:
I think it is ok. Please reopen it if needed.

closed 2015-03-27 by agdl

cmaglie, 2015-04-22:
I've integrated the description by @agdl with some more details.

https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5---3rd-party-Hardware-specification#build-process
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5---3rd-party-Hardware-specification#boardstxt

here the diff:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5---3rd-party-Hardware-specification/_compare/9574fc76da1ce0fd5c96364f4293ed68b2f7de25...b16d462af0da1b6e0c59b8a0879db64a2694eefb