arduino/Arduino #2649: Cannot set Leonardo extended fuses to 0xCB

NicoHood, 2015-02-16:
The Arduino IDE wont upload this for 32u4 chips but for 16u2 chips.

```
C:\Users\Nico\Documents\arduino-1.6.0/hardware/tools/avr/bin/avrdude -CC:\Users\Nico\Documents\arduino-1.6.0/hardware/tools/avr/etc/avrdude.conf -v -patmega32u4 -carduino -PCOM40 -b19200 -e -Ulock:w:0x3F:m -Uefuse:w:0xFC:m -Uhfuse:w:0xD8:m -Ulfuse:w:0xFF:m 

avrdude: Version 6.0.1, compiled on Apr  3 2014 at 23:12:16
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "C:\Users\Nico\Documents\arduino-1.6.0/hardware/tools/avr/etc/avrdude.conf"

         Using Port                    : COM40
         Using Programmer              : arduino
         Overriding Baud Rate          : 19200
         AVR Part                      : ATmega32U4
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  9000  9000 0x00 0x00
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 2
         Firmware Version: 1.18
         Topcard         : Unknown
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e9587
avrdude: erasing chip
avrdude: reading input file "0x3F"
avrdude: writing lock (1 bytes):

Writing | ################################################## | 100% 0.02s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0x3F:
avrdude: load data lock data from input file 0x3F:
avrdude: input file 0x3F contains 1 bytes
avrdude: reading on-chip lock data:

Reading | ################################################## | 100% 0.02s

avrdude: verifying ...
avrdude: 1 bytes of lock verified
avrdude: reading input file "0xFC"
avrdude: writing efuse (1 bytes):

Writing |  ***failed;  
################################################## | 100% 0.16s

avrdude: 1 bytes of efuse written
avrdude: verifying efuse memory against 0xFC:
avrdude: load data efuse data from input file 0xFC:
avrdude: input file 0xFC contains 1 bytes
avrdude: reading on-chip efuse data:

Error while burning bootloader.
Reading | ################################################## | 100% 0.02s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0000
         0xcc != 0xfc
avrdude: verification error; content mismatch

avrdude done.  Thank you.

```

NicoHood, 2015-02-16:
The funny thing is i cannot set the 16u2 fuses to 0xCB.

To summarize:

chip | working | not working
--- | --- | ---
32u4 | 0xFC | 0xCB
16u2 | 0xCB | 0xFC

Fede85, 2015-07-01:
Hi Nico,

I've checked in the datasheet and for both the ATmega16U2 (p.247) and the ATmega32U4 (p.347) the bits 4..7 of the extended fuse are unused/reserved and set to value 1 by default for both the microcontrollers. In my experience they can be programmed once, when a 0 is written in these bits it's permanent and can be fixed only with an high-voltage programmer.

So, the high nibble should be set to 0xF_ for consistency but practically you can write any value and in this case when you write 0xC_ you have to continue to use this value, for example if your extended fuse is set to 0xCB then you can change it to 0xCC but not to 0xFC if you don't have an high voltage programmer.

f

matthijskooijman, 2015-07-02:
What programmer is being used here? There is an issue with avrdude, where depending on the programmer used, unused fuse bits are verified as 0 or 1. I believe that some programmers leave them at 0, but some just return the 1s returned by the hardware. See https://savannah.nongnu.org/bugs/?21954

NicoHood, 2015-08-17:
I used the arduino as ISP with the arduino (not stk500v1) setting and bruned via IDE.

BrentWilkins, 2015-09-01:
I can reproduce the same thing here.  The IDE gives up on the mismatch.  If I run the same avrdude command from the command line, avrdude offers to fix the extended fuses.  In my case it tried 10 times and failed.  Used at Atmel-ICE & the bits flipped as desired on the first try.

per1234, 2017-07-03:
Using avrdude 6.3.0-arduino9 I get the following when I set extended fuse on ATmega32U4 to 0xFC using Arduino as ISP:
```
E:\ArduinoIDE\arduino-1.8.3\hardware\tools\avr/bin/avrdude -CE:\ArduinoIDE\arduino-1.8.3\hardware\tools\avr/etc/avrdude.conf -v -patmega32u4 -cstk500v1 -PCOM40 -b19200 -e -Ulock:w:0x3F:m -Uef
use:w:0xfc:m -Uhfuse:w:0xd8:m -Ulfuse:w:0xff:m

avrdude: Version 6.3, compiled on Jan 17 2017 at 12:00:53
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "E:\ArduinoIDE\arduino-1.8.3\hardware\tools\avr/etc/avrdude.conf"

         Using Port                    : COM40
         Using Programmer              : stk500v1
         Overriding Baud Rate          : 19200
         AVR Part                      : ATmega32U4
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  9000  9000 0x00 0x00
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : STK500
         Description     : Atmel STK500 Version 1.x firmware
         Hardware Version: 2
         Firmware Version: 1.18
         Topcard         : Unknown
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.06s

avrdude: Device signature = 0x1e9587 (probably m32u4)
avrdude: safemode: hfuse reads as D8
avrdude: safemode: efuse reads as CB
avrdude: erasing chip
avrdude: reading input file "0x3F"
avrdude: writing lock (1 bytes):

Writing | ################################################## | 100% 0.02s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0x3F:
avrdude: load data lock data from input file 0x3F:
avrdude: input file 0x3F contains 1 bytes
[... 40 lines cut ...]
avrdude: reading input file "0xff"
avrdude: writing lfuse (1 bytes):

Writing | ################################################## | 100% 0.01s

avrdude: 1 bytes of lfuse written
avrdude: verifying lfuse memory against 0xff:
avrdude: load data lfuse data from input file 0xff:
avrdude: input file 0xff contains 1 bytes
avrdude: reading on-chip lfuse data:

Reading | ################################################## | 100% 0.02s

avrdude: verifying ...
avrdude: 1 bytes of lfuse verified

avrdude: safemode: hfuse reads as D8
avrdude: safemode: efuse reads as CC
avrdude: safemode: Fuses OK (E:FC, H:D8, L:FF)

avrdude done.  Thank you.
```
So you can see that the fuse still reads as 0xCC but now the process does complete successfully and there is a warning that explains the cause of the discrepancy. The results are the same with USBasp so it's not isolated to Arduino as ISP. I'm thinking this issue can be closed, do you agree @NicoHood?

Note that I think the table in https://github.com/arduino/Arduino/issues/2649#issuecomment-74528770 has the fuse values swapped.

closed 2018-04-28 by per1234