arduino/Arduino #2653: 1.6.0 IDE sets "file modified" flag whenever a tab is clicked for viewing, instead of when the content is actually changed

plocher, 2015-02-16:
Context:
* MacOS
* Arduino IDE 1.6.0

To reproduce:
* Start with a multi-tab sketch
* Save it (ensures all "file modified" state/markings are cleared for this example...)
* Click on another tab but do not modify its content
* Notice the tab now shows the "file modified" squiggle even though nothing was done to modify the content of the tab

IMO, the "modified" state should track the changelog/undo buffer, such that "undo"ing all edits clears the modified flag and the modified state only is set after a true content modification.  The current implementation seems to be much less nuanced :-)

ffissore, 2015-02-19:
Can reproduce this on linux. @cmaglie may you test this with your mac?

cmaglie, 2015-02-19:
Happens also on linux, I can reproduce it with the following sequence:

* Open a sketch
* *Modify the sketch and save* (add some empty lines for example)
* Create a new tab, for example "test.h"
* Save
* Click on the first tab
* Now the first tab is marked as modified

ffissore, 2015-02-19:
A fix will be available at #2666 in the next couple of hours

ffissore, 2015-02-24:
@plocher can you give a spin to https://github.com/arduino/Arduino/pull/2666#issuecomment-75529858 and report if it fixes your issue?

plocher, 2015-02-24:
> @plocher <https://github.com/plocher> can you give a spin to #2666
> (comment)
> <https://github.com/arduino/Arduino/pull/2666#issuecomment-75529858> and
> report if it fixes your issue?

Will try to find time in midst of business travel tonight 🌃

PR #2666 (2015-03-01): Undo sets the sketch to modified, even if it's not
Fixes #2653
/cc @plocher @cmaglie
commit 78f2e87b (2015-03-01): Undo and Redo state update were setting the sketch to modified, even if it was not. Fixes #2653
closed 2015-03-01 by ffissore