arduino/Arduino #2655: "Save as" should default to parent of current sketch folder.

damellis, 2015-02-16:
On Mac at least, when you select "Save as..." from the File menu, the dialog box has the sketch's folder selected by default. That means if you simply type a new name and press enter, you'll get an error about trying to save a copy of the sketch inside itself ("How very Borges of you").

It would probably make more sense for the dialog box to default to the parent of the current sketch folder instead, i.e. default to saving the copy next to the original instead of inside it.

This is with 1.6 (the recommended version) on Mac OS X 10.10.2.

ivovdgrift, 2015-02-27:
It seems to only do this on mac, not on windows. I'll try to create a fix for macs

ivovdgrift, 2015-02-27:
Offered a solution [here](https://github.com/arduino/Arduino/pull/2716)

ffissore, 2015-03-04:
Fixed with #2716

closed 2015-03-04 by ffissore